wpilibsuite/allwpilib #2590: notifierCommandScheduleTest is Unstable

AustinShalit, 2020-07-12:
```Task :wpilibNewCommands:test
edu.wpi.first.wpilibj2.command.NotifierCommandTest > notifierCommandScheduleTest() FAILED

    org.opentest4j.AssertionFailedError: Should have hit at least 10 triggers ==> expected: <true> but was: <false>
        at org.junit.jupiter.api.AssertionUtils.fail(AssertionUtils.java:39)
        at org.junit.jupiter.api.AssertTrue.assertTrue(AssertTrue.java:40)
        at org.junit.jupiter.api.Assertions.assertTrue(Assertions.java:193)
        at edu.wpi.first.wpilibj2.command.NotifierCommandTest.notifierCommandScheduleTest(NotifierCommandTest.java:32)```

calcmogul, 2021-07-11:
Have we hit this error at all recently?

calcmogul, 2021-08-30:
All recent test failures are for things besides this. If it comes up again, this issue should be reopened.

closed 2021-08-30 by calcmogul