wpilibsuite/allwpilib #2601: HALSIM_RegisterDriverStationAllCallbacks doesn't notify on Joystick data update

virtuald, 2020-07-18:
**Is your feature request related to a problem? Please describe.**
For websocket. Want notification of all changes.

**Describe the solution you'd like**
Gimme notifications.

**Describe alternatives you've considered**
Crying.

amorygalili, 2020-07-18:
For the websim I'm making using Dustin's websocket as the back end, this notification is needed to properly update the front end.

PeterJohnson, 2020-07-21:
What are you expecting the notifications to contain?  Joystick data is quite complex - e.g. there's sized arrays of axes, POVs, and buttons, as well as outputs.  Do you just want a notification when any of that changes in any way for any joystick, and then you'll grab the data and compute deltas from it?

PR #2628 (2020-08-04): [sim] Add callbacks for joysticks and match info
Fixes #2601.
closed 2020-08-04 by PeterJohnson