wpilibsuite/allwpilib #2619: Create GitHub Actions repository

Starlight220, 2020-07-29:
Create a repository of GitHub Actions `*.yml` files useful for teams, so teams can easily set up CI/CD in their repository by copying `*.yml` files from there.

Having checkstyle/wpiformat there would be a good start, then possibly something a vendordep one once wpilibsuite/gradleRIO#418 is merged.

Starlight220, 2020-07-29:
Another idea is adding the robot project templates as GitHub template repositories, then teams that use git can have an easier setup, including not having to use VSCode.

prateekma, 2020-07-29:
It would also be nice if something like the snippet below was supported:
```yml
- uses: wpilibsuite/check-wpiformat@master
  name: Check WPIFormat Output
```

Starlight220, 2020-07-29:
Doing this will also facilitate wpilibsuite's transition to Github Actions from Azure Pipelines.

From a quick reading of the [docs](https://docs.github.com/en/actions/creating-actions/publishing-actions-in-github-marketplace), it seems that each action needs a separate repository. I can do most of the work needed with some help, given that I would have the needed permissions.

Starlight220, 2020-08-03:
Where is wpiformat developed? I didn't see a repo for it under the wpilibsuite org.

prateekma, 2020-08-03:
https://github.com/wpilibsuite/styleguide

Starlight220, 2020-08-20:
I think this can be closed: discussion showed that wpiformat doesn't help most teams, and what other Actions would be useful for most teams?

closed 2020-09-08 by AustinShalit