wpilibsuite/allwpilib #2620: ValueError('Unable to configure handler ' ValueError: Unable to configure handler 'file_handler'

kirupakarankumar, 2020-07-29:
I am deploying python script as docker container in GKE.
Triggered as cronjob & setting logs to mount on some path.

Getting below error.

Traceback (most recent call last): File "open_src_js.py", line 255, in <module> main() File "open_src_js.py", line 229, in main set_up_logging(options) File "open_src_js.py", line 221, in set_up_logging set_logging(console_log_level=options.file_log_level, file_log_level=options.file_log_level, File "/workspace/src/json_logger.py", line 65, in set_logging logging.config.dictConfig(conf) File "/usr/local/lib/python3.8/logging/config.py", line 808, in dictConfig dictConfigClass(config).configure() File "/usr/local/lib/python3.8/logging/config.py", line 570, in configure raise ValueError('Unable to configure handler ' ValueError: Unable to configure handler 'file_handler'

Can anyone help on this ??

Kevin-OConnor, 2020-07-29:
Hi there, it doesn't seem like any of these messages or your issue description are related to this project (WPILib), maybe you posted in the wrong place?

AustinShalit, 2020-07-29:
I think you opened this issue on the wrong repository. This repository contains the source code for FIRST Robotics Competition teams. I'm not sure what software that error comes from but I would find its repository using the search bar and open an issue for it.

closed 2020-07-29 by AustinShalit

kirupakarankumar, 2020-07-29:
sorry for that