wpilibsuite/allwpilib #2642: HAL_GetPDPTotalEnergy computation appears bogus

chauser, 2020-08-17:
**Describe the bug**
The computation in `HAL_GetPDPTotalEnergy` appears bogus. It introduces a local variable, `energyJoules`, initializes it from another local variable, `raw`, updates it with several computations, then returns the same expression of `raw` that was used to initialize `totalEnergy`. Without know the details of what it is supposed to do it is hard to say it is out and out wrong, but if it is right it is doing unnecessary computation. 

```
double HAL_GetPDPTotalEnergy(HAL_PDPHandle handle, int32_t* status) {
  PdpStatusEnergy pdpStatus;
  int32_t length = 0;
  uint64_t receivedTimestamp = 0;

  HAL_ReadCANPacketTimeout(handle, StatusEnergy, pdpStatus.data, &length,
                           &receivedTimestamp, TimeoutMs, status);
  if (*status != 0) {
    return 0;
  }

  uint32_t raw;
  raw = pdpStatus.bits.Energy_125mWPerUnitXTmeas_h4;
  raw <<= 8;
  raw |= pdpStatus.bits.Energy_125mWPerUnitXTmeas_mh8;
  raw <<= 8;
  raw |= pdpStatus.bits.Energy_125mWPerUnitXTmeas_ml8;
  raw <<= 8;
  raw |= pdpStatus.bits.Energy_125mWPerUnitXTmeas_l8;

  double energyJoules = 0.125 * raw; /* mW integrated every TmeasMs */
  energyJoules *= 0.001;             /* convert from mW to W */
  energyJoules *=
      pdpStatus.bits
          .TmeasMs_likelywillbe20ms_; /* multiplied by TmeasMs = joules */
  return 0.125 * raw;
}
```

PeterJohnson, 2020-08-17:
The raw computation is needed to combine the bits from different status parts into a single 32-bit value.

chauser, 2020-08-17:
Yes, but that's not what I'm talking about. The subsequent computation of
totalEnergy is never used.

chauser, 2020-08-17:
Sorry, I should have said the computation of energyJoules is not used.

ThadHouse, 2020-08-17:
Yup, its 100% a bug.

https://github.com/wpilibsuite/allwpilib/blob/e21a246a4d9b7bcaf931a4d785fbf9193fdaf3d5/hal/src/main/native/athena/ctre/PDP.cpp#L181

This was the code before the rewrite

PR #2672 (2020-08-31): [hal] Fix PDP Energy Calculation
Fixes #2642
commit ad6c8b88 (2020-08-31): [hal] Fix PDP Energy Calculation (#2672)

Fixes #2642
closed 2020-08-31 by PeterJohnson