wpilibsuite/allwpilib #2664: Provide hook for error object reporting

virtuald, 2020-08-28:
In C++, when something like a duplicate AnalogOutput is created, `ErrorBase::SetWPIError` is called, which merely prints a message to the console that is easily ignored. I believe the intent was to avoid C++ exceptions because of poor debugging experience in C++? 

Java throws an exception. RobotPy would like to throw an exception, but there's currently no easy way to override the error reporting mechanism.

It seems the least invasive way to allow this would be to provide some static function in `Error` that would override the default reporting mechanism (currently DriverStation::ReportError). Currently report is only called periodically, it would be nice if the report period could be overridden as well.

I'm open to other suggestions as well.

Reference: https://github.com/robotpy/robotpy-wpilib/issues/633

ThadHouse, 2020-08-28:
I'm actually of the mindset we should both do this for runtime errors, but also turn errors into exceptions in constructors. A failing construction is unrecoverable imo, and should crash the code unless the user explicitly catches it. Our HAL objects are rarely multiple initialized on purpose, so its basically always an unrecoverable user error.

ThadHouse, 2020-08-28:
I was going to propose doing that for 2022. I know we'd had discussions on it in the past that really just kind of ended.

virtuald, 2020-08-28:
As mentioned above, I feel there was opposition to exceptions in the past? But might be a good idea for 2022. 

I would be satisfied with a hook for 2021. :-D

ThadHouse, 2020-08-28:
The opposition was mostly in doing it for runtime errors. Construction didn't have as much opposition because you're dead anyway without recourse. And if you're doing it on purpose, you can catch.

calcmogul, 2020-08-28:
In the state-space classes, we've been calling ReportError() then throwing exceptions in ctors when the user gives bad arguments. Having a zombie object is harder to notice and fix than the code being in a reboot loop with the error message for it spamming the console.

PeterJohnson, 2021-04-19:
Essentially fixed by #3306, which changes to exceptions for hard failures.

closed 2021-04-19 by PeterJohnson