wpilibsuite/allwpilib #2694: Static HTTP response to websocket clients in ntcore

Redrield, 2020-09-06:
Per spec discussion, consider adding a case to ntcore to detect clients attempting to connect to TCP 1735 over websocket, and to redirect them to the port dedicated to serving websocket clients (Should multiplexing not be supported in the currrent ntcore server).

An example of how to discriminate WS packets can be found [here](https://github.com/first-rust-competition/nt-rs/blob/master/src/proto/server/conn.rs#L36), checking the first 3 bytes for the ascii "GET" will identify a client attempting to establish a websocket connection, and in that case the server can return a static HTTP response, either denying the client outright with a reason in the body, or redirecting to the port handling websocket clients. (nt-rs returns HTTP 405 when not configured for websocket clients, in the case where the server isn't configured to multiplex on one port, HTTP 301 can be used to redirect clients)

Redrield, 2020-09-07:
In doing some testing, it seems like a redirect wouldn't work for most client implementations. The Autobahn test suite clients just hang forever, and the library I use in Rust throws an error when the response code isn't HTTP 101.

PeterJohnson, 2021-12-31:
Closing as NT4 (WS-based) will use a different port.

closed 2021-12-31 by PeterJohnson