wpilibsuite/allwpilib #2701: Add a method for not starting LiveWindow in IterativeBase

sciencewhiz, 2020-09-12:
**Is your feature request related to a problem? Please describe.**
LiveWindow always starts in test mode. This may interfere with code that the user wants to run in test mode. There is no way to keep it from starting. The user can disable it in testInit, but this is after it starts.

**Describe the solution you'd like**
A method of disabling LiveWindow so it isn't started at all.

Starlight220, 2020-09-12:
Where would teams call this function, if LW starts before `testInit()`? How does LW interfere with user code?

sciencewhiz, 2020-09-12:
> Where would teams call this function, if LW starts before testInit()?

RobotInit, or anywhere else the user wants before testInit. For example, they might want to only disable it sometimes based on user input.

> How does LW interfere with user code?

Here's examples of LiveWIndow interfering with what people want to do in Test Mode.
https://www.chiefdelphi.com/t/wpilib-blog-wpilib-2020-2-2-update/373295/8
https://www.chiefdelphi.com/t/can-you-run-commands-in-test-methods-of-robot-java/349402
https://www.chiefdelphi.com/t/testinit-does-not-appear-to-issue-commands-in-simulator/387753

Starlight220, 2022-10-02:
In #3217, LW is disabled by default.

sciencewhiz, 2022-10-02:
> In #3217, LW is disabled by default.

Where? It's still enabled here: https://github.com/wpilibsuite/allwpilib/blob/291b2ec05df62ea7d3571d8eebf47d21bf849428/wpilibj/src/main/java/edu/wpi/first/wpilibj/IterativeRobotBase.java#L296-L301

Starlight220, 2022-10-02:
My mistake.