wpilibsuite/allwpilib #2715: Hardcode SwerveKinematics to 4 modules

calcmogul, 2020-09-15:
Removing the templatization would simplify the internals a lot. The only
swerve drives we're aware of that haven't had four wheels are FRC team
1625 in 2010 (6 wheels) and FRC team 179 in 2019 (3 wheels).

calcmogul, 2020-09-15:
Since everything uses parameter packs for arguments, we can just take four arguments instead. The only user-facing breakage I see for 4-wheel swerve teams is having to remove `<4>` from a declaration of SwerveDriveKinematics as a member variable (i.e., `SwerveDriveKinematics<4>`).

sciencewhiz, 2020-09-15:
I think there's been more 3 wheel swerves. I think 16 has done it. Team 67 in 2005. 

However, with light brushless motors and a 20 port PDP, I think 3 wheel swerves are even less likely going forward.

auscompgeek, 2021-02-27:
My 2c: 4774 usually does testing with 2 wheel swerve for proof of concepts.

AustinShalit, 2021-03-12:
This class works as is and we do not feel strongly about lowering the roof.

closed 2021-03-12 by AustinShalit