wpilibsuite/allwpilib #2718: Migrate usages of System.out.println() to DriverStation.reportWarning()

Starlight220, 2020-09-17:
https://github.com/wpilibsuite/allwpilib/blob/56972447b22d8f9fcb65e1f650c8c824377076d4/wpilibNewCommands/src/main/java/edu/wpi/first/wpilibj2/command/CommandScheduler.java#L326
In this usage specifically, the header and the rest of the message are printed separately, and I even think that the rest of the message is printed to `DriverStation.reportWarning()`. This way the text might be split around other printed text. The lambda overload of `Tracer.printEpochs()` exists for that reason.

calcmogul, 2020-09-17:
Yea, the C++ and Java versions of CommandScheduler should be doing what IterativeRobotBase does.

Watchdog doesn't inherit from Tracer because the APIs don't match. Tracer has public methods that don't make sense for Watchdog to expose like ClearEpochs() (that function is called automatically when appropriate by Watchdog).

Starlight220, 2020-09-17:
So I'm submitting a PR that adds the lambda overload of `Tracer.printEpochs()` to the `Watchdog` class. And when that is merged I'll fix the sysout usages.

Starlight220, 2022-05-02:
I think this can be closed?

closed 2022-05-02 by Starlight220