wpilibsuite/allwpilib #2731: Implement full HAL support in WebSocket extension

PeterJohnson, 2020-09-22:
While the WebSocket extension has decent coverage of the HAL, it is currently missing support for the following HAL features:
- [ ] Accelerometer
- [ ] AddressableLED
- [ ] AnalogGyro
- [ ] Compressor
- [ ] PDP
- [ ] Solenoid
- [ ] Timing (maybe not worth it?)

It would be great if those could get implemented as well (it's mostly just more boilerplate in halsim_ws_core).

PeterJohnson, 2020-09-22:
@Redrield

bvisness, 2020-12-05:
Not having Solenoid is turning out to be a real bummer for the simulation stuff we're currently trying to build as prep for the season. Is @Redrield still around to wrap up the existing work? Alternatively, how hard would it be for me to add Solenoid specifically for our own purposes? (I could contribute that little bit back, at least.)

zhiquanyeo, 2020-12-05:
It's not too hard to add a new provider. (See https://github.com/wpilibsuite/allwpilib/blob/master/simulation/halsim_ws_core/src/main/native/cpp/WSProvider_DIO.cpp as an example)

There might be some small changes to the WS message field names, but for the most part, the work consists of hooking into the HALSIM callbacks