wpilibsuite/allwpilib #2761: SplineHelper.getQuinticSplinesFromControlVectors() requires an even number of control vectors

CarlinWilliamson, 2020-10-02:
**Describe the bug**
[`SplineHelper.getQuinticSplinesFromControlVectors()`](https://github.com/wpilibsuite/allwpilib/blob/330b90e046ae12c88d4ad86a5be10771232a2f57/wpimath/src/main/java/edu/wpi/first/wpilibj/spline/SplineHelper.java#L217) contains an integer division by 2. Consequently passing an odd number of control vectors will return a spline which does not include the final point.

**To Reproduce**
Pass in a three control vector list into `SplineHelper.getQuinticSplinesFromControlVectors()`.
The generated spline will stop at the second control vector.

**Expected behavior**
The spline should start at the first vector, pass through the second vector and finish at the third vector

**Desktop (please complete the following information):**
 - OS: ubuntu 20.04
 - Java version: openjdk 11.0.8

**Additional context**
The expected behaviour can be achieved by passing dupliating intermediary control vectors (not the start or end)
I.e. you would need to pass in [start, A, A, end] if you want a spline which passes through control vector A

prateekma, 2020-10-02:
Good catch. The `quinticControlVectorsFromWaypoints` intentionally generates duplicate control vectors so this isn't an issue when teams are using `TrajectoryGenerator` interface.

However, I assume PathWeaver will want to call this method directly, so I'll see if I can reorganize the `SplineHelper` code in a way that doesn't require duplicated control vectors.

CarlinWilliamson, 2020-10-02:
Haha. That's exactly what I'm working on. Thanks!

PR #2762 (2020-10-04): [wpimath] Fix quintic spline generation from control vectors
Closes #2761.

This PR does not introduce any breaking changes for teams that used the TrajectoryGenerator API for quintic splines with poses. 

The `GetQuinticControlVectorsFromWaypoints()` method was removed because it is not possible for us (or would require breaking changes to the shape of the splines) to generate only one quintic control vector per Pose2d (https://github.com/wpilibsuite/allwpilib/pull/2429) whereas users who *actually have control vectors directly* (i.e. not from Pose2d objects, but a dashboard such as PathWeave [... cut]
closed 2020-10-04 by PeterJohnson