wpilibsuite/allwpilib #2792: PIDSubsystem doesn't run in test mode

sciencewhiz, 2020-10-21:
In the old command framework, PIDSubsystem's PID controller would run in test mode in its own thread, allowing the controllers to be tuned using LiveWindow while in test mode.
In the new command framework, PIDSubsystem's PID controller is run in the periodic method of the subsystem. The command scheduler is disabled in test mode, so the periodic method isn't run, and thus the PID Controller isn't run, and while the PID constants and setpoint can be changed in test mode, the robot has to be enabled in teleop in order to run the pid controller.

Starlight220, 2020-10-21:
How exactly is `CommandScheduler` disabled in Test mode? It runs whenever `.run()` is called, and in worst case `.enable()` can be called.

sciencewhiz, 2021-01-15:
It's disabled here: https://github.com/wpilibsuite/allwpilib/blob/406d055f076bf910c691c798c5b2e750e6f1f2dd/wpilibNewCommands/src/main/java/edu/wpi/first/wpilibj2/command/CommandScheduler.java#L85-L97

PeterJohnson, 2022-03-29:
I would prefer to narrowly define LiveWindow behavior for test mode, and communicate to users that things like PID tuning should be done in other modes (e.g. teleop with a separate joystick port) unless they want to disable LiveWindow test mode completely.  Given that, I think the right answer here is to simply document how to change test mode to use it for something other than LiveWindow.

Starlight220, 2022-10-02:
Related to #2701, and would probably be fixed by the same change.

sciencewhiz, 2022-10-02:
While disabling livewindow like in #2701 would allow the PID controller to run, you don't get the other benefits of livewindow, which the beta tester wanted. 

This fix would be more involved and probably be something like runWhenTest like we have for disabled.

Starlight220, 2022-10-02:
I don't think another `runsWhenDisabled` is a good idea or solution for this.

There are multiple options, all of them currently usable:
- Cutting out the scheduler: calling the subsystem's `periodic` or command lifecycle method(s) directly. This might be unsafe if some library or team-written types make certain assumptions, but there's not much to do about that.
- `CommandScheduler.enable()`. If `run()` is called from `robotPeriodic` then this will work as long as LW and whatever code the scheduler runs don't interfere with each other's hardware writes.
- Disabling LW. This will automatically enable the scheduler.

I think that documenting the above options and perhaps changing some defaults (like whether LW is enabled in test mode) would be more than enough.

Starlight220, 2023-03-02:
An idea: if LW control can be made more granular, the scheduler can treat LW as a command that requires subsystems controlled by LW, rather than completely disabling.

truher, 2023-11-19:
In case anyone is monitoring this issue, I wanted to share my experience of the past little while, during which i discovered the consequences of making a category error about LiveWindow.  Given the plethora of dashboard options available, I thought that choosing one and ignoring the others would be fine, so we chose Glass and ignored everything else, including LiveWindow, and we also considered the very idea of a "dashboard" to be about passive observation, which is our main use case.

So today, working on testing, it was a struggle to discover why the command scheduler wasn't running in test mode.  Isn't test mode just a different set of init, periodic, and exit functions? ... NO!   Test mode involves a specific feature of a specific dashboard, all of which we have completely ignored until now, that turns off the command scheduler.  So the category error is considering LiveWindow to be a dashboard feature.  LiveWindow is not a passively-observing dashboard after all, it is in a sense the "master" of a very core component, the scheduler.

IMHO it would be better for "robot stuff" and "dashboard stuff" to be more decoupled at the framework level.  If you want to enable or disable the command scheduler in testInit(), you can.  If you want to use any dashboard to look at stuff while the command scheduler is disabled, you can do that too.  But wiring this stuff together in way that's kinda hard to discover doesn't seem great.

Anyway, I eventually discovered enableLiveWindowInTest(), which solves my immediate problem, but I thought someone might be interested in the feedback about the design issues here.  Thanks for listening.  :-)

Starlight220, 2023-11-20:
What action items do we want in order to resolve this issue?

closed 2024-01-15 by Starlight220