wpilibsuite/allwpilib #2801: Improve assertions

PeterJohnson, 2020-10-23:
Currently we use plain assert() in various places.  This includes in imgui, which uses plain assert() because we don't define IM_ASSERT.  Unfortunately plain assert() is not compiled in release builds, which is what we use for JNI libraries.  For our use case, it would be much better if we globally used our own assertion function which is compiled even into release builds.  There is a wpi_assert() in wpilibc (Utility.h), but it's a bit inefficiently written (it checks the condition inside a function call, so you always pay the function call penalty).  It also does not end the program but simply reports an error and continues.  Recommend doing the following:

- [ ] Change wpi_assert to end the program instead of continuing.  There are only a couple of places in wpilibc which should actually continue on error--change these to do "normal" error reporting.  These are also the only locations that use the fact wpi_assert returns a boolean, so remove that functionality as well (void return).
- [ ] Move wpi_assert() and kin to wpiutil
- [ ] Optimize wpi_assert() by checking the condition in the macro (or an inline C++ function it calls) and only reporting in the actual function.  Change parameters back to const char* to reduce header burden.
- [ ] Define IM_ASSERT in imgui to use wpi_assert
- [ ] Assess other locations where an assert should be used in unrecoverable fail-fast situations and add wpi_assert() there.

virtuald, 2020-10-23:
Generally asserts are intentionally debug-build only things. If you want release build checks maybe give it a different name?

PeterJohnson, 2020-10-23:
The problem is that we always use release builds in some situations.  The intent is the same as an assert... “this should never happen” things.  It’s just better to print a message and fail fast on a nullptr than just let things crash and get a JVM crash message that doesn’t help us.  We can use a different name than assert (although wpi_assert *is* a different name)... I think windows uses “verify” in a few places for similar things?

virtuald, 2020-10-24:
'verify' or 'check' seem more appropriate than assert.

ThadHouse, 2020-10-24:
`wpi_yeet`?

PeterJohnson, 2020-10-24:
I guess the issue I have with "verify" or "check" is that doesn't scream to me "this aborts the entire program if the condition is untrue".  "assert" does.

auscompgeek, 2020-10-24:
We could consider other prior art here. Rust has both a `debug_assert!` and an `assert!`; the former only runs in a debug build, the other runs unconditionally.

PeterJohnson, 2020-10-24:
`wpi_release_assert`?  A little long but workable.

calcmogul, 2021-07-11:
OBE by #3306 and #3376.

closed 2021-07-11 by calcmogul