wpilibsuite/allwpilib #2807: How to Change C++ Compiler in Windows

rsammelson, 2020-10-25:
Currently gradle is trying to use the Visual Studio 2015 compiler, which does not support C++ 17. Instead I get the error `cl : Command line warning D9002 : ignoring unknown option '/std:c++17'`.

I have the Visual Studio 2019 toolchain installed, so how can I switch to that compiler?

ThadHouse, 2020-10-25:
Gradle actually determines this internally, and we don't really have a way to control it that I know of. I'll look into it though. 

When you say you have 2019 installed, do you just mean the 2019 build tools, or full vs 2019? Gradle only sees full vs installs.

rsammelson, 2020-10-25:
Oh, I just have the build tools. What was the WPILib installer supposed to install, since it has a box for C++ stuff?

ThadHouse, 2020-10-25:
The wpilib installer just installs the roboRIO c++ compiler. It doesn't install the desktop compiler, that has to be installed separately.

rsammelson, 2020-10-25:
Installing the full VS appears to have fixed the issue; thanks!

closed 2020-10-25 by AustinShalit