wpilibsuite/allwpilib #2809: Build error: PROTOBUF_ARRAY_SIZE

coderkevin, 2020-10-25:
**Describe the bug**
I'm getting a consistent build on current master when I try to build basically anything:

```
compiling float64.pb.cc failed.
In file included from /Users/kevinkillingsworth/projects/robotics/allwpilib/simulation/gz_msgs/build/generated/main/simulation/gz_msgs/float64.pb.cc:4:
In file included from /Users/kevinkillingsworth/projects/robotics/allwpilib/simulation/gz_msgs/build/generated/main/simulation/gz_msgs/float64.pb.h:26:
In file included from /usr/local/include/google/protobuf/generated_message_table_driven.h:35:
In file included from /usr/local/include/google/protobuf/map_entry_lite.h:42:
/usr/local/include/google/protobuf/generated_message_util.h:230:14: error: zero size arrays are an extension [-Werror,-Wzero-length-array]
  void* deps[PROTOBUF_ARRAY_SIZE(N)];
             ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/google/protobuf/generated_message_util.h:215:32: note: expanded from macro 'PROTOBUF_ARRAY_SIZE'
#define PROTOBUF_ARRAY_SIZE(n) (n)
                               ^~~
/Users/kevinkillingsworth/projects/robotics/allwpilib/simulation/gz_msgs/build/generated/main/simulation/gz_msgs/float64.pb.cc:36:47: note: in instantiation of template class 'google::protobuf::internal::SCCInfo<0>' requested here
::PROTOBUF_NAMESPACE_ID::internal::SCCInfo<0> scc_info_Float64_float64_2eproto =
                                              ^
1 error generated.

compiling bool.pb.h successful.
```

**To Reproduce**
Steps to reproduce the behavior:
Using OSX Catalina 10.15.4
1. `./gradlew build` and this happens several times

**Expected behavior**
A successful build

**Screenshots**
See above compiler output

**Desktop (please complete the following information):**
 - OS: Mac OSX Catalina 10.15.4

```
$ java --version
java 13.0.1 2019-10-15
Java(TM) SE Runtime Environment (build 13.0.1+9)
Java HotSpot(TM) 64-Bit Server VM (build 13.0.1+9, mixed mode, sharing)
```

```
$ gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.3 (clang-1103.0.32.62)
Target: x86_64-apple-darwin19.4.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

prateekma, 2020-10-25:
@coderkevin Can you try out #2810 and see if that fixes the issue?

sciencewhiz, 2020-10-27:
What version of protobuf do you have installed?

coderkevin, 2020-11-01:
> What version of protobuf do you have installed?

3.13.0_1

Is there a better version for compatibility?

coderkevin, 2020-11-01:
> @coderkevin Can you try out #2810 and see if that fixes the issue?

I tried it, and the build does succeed now. However, I think this issue should remain open to resolve the protobuf dependency issue.

ThadHouse, 2020-11-01:
We've never tested Gazebo and Protobuf support on macos, and we don't really support it on linux either. Its not maintained, and kind of just there in case someone gets to it in the future.

coderkevin, 2020-11-01:
Good to know, thanks. In that case I think this can be closed when #2810 merges then.

prateekma, 2020-11-01:
Even when sim is enabled on macOS, the failure is due to Clang not liking zero length arrays. I've also disables that warning for the gz_msgs build in #2810.

PR #2810 (2020-11-09): [build] Disable Gazebo builds when -PmakeSim is not set
Closes #2809
closed 2020-11-09 by AustinShalit