wpilibsuite/allwpilib #2814: Livewindow analog potentiometer scale

cpapplefamily, 2020-10-26:
Yeas ago the analog potentiometer displayed the user units as entered in the range field. The PID controller uses this value but the livewindow scales the input for 0 to 5 now. Getting this display to scale as the entered range would help with PID tuning.

sciencewhiz, 2020-10-26:
This appears to have changed behavior in wpilibsuite/allwpilib#771. Previously it returned the scaled value from AnalogPotentiometer, and after it returns the analog input value.

cpapplefamily, 2020-10-27:
Well I "Hacked" In a copy of the AnalogPotentiometer.Java class into my project and imported my instance of the class with this edit:
```
  @Override
  public void initSendable(SendableBuilder builder) {
    if (m_analogInput != null) {
      //m_analogInput.initSendable(builder);
      builder.setSmartDashboardType("Analog Input");
      builder.addDoubleProperty("Value", this::get, null);
    }
  }
```

yep the LiveWindow returns my scaled value only the widget needs to be manually rescaled too. maybe `builder.setSmartDashboardType` needs to accept more varialbes to scale the "Analog Input"? Is this a function that can be reverted back? Or will it break other things I obviously no nothing about?

sciencewhiz, 2020-10-27:
Best I can tell, the min and max levels were never published, nor did smartdashboard or shuffleboard read them from network tables.

cpapplefamily, 2020-10-27:
Maybe I just looked at the Value in the NetworkTables, or back in the SmartDashboard days it looks like it used to simply display the value as text not as a slider. Either way If the above changes don't effect other aspects of how things work i might put a Pull Request together. Instead of using a Dashboard Type " Analog Input" I'll look for a string display to show the scaled Value.

closed 2020-11-02 by cpapplefamily