wpilibsuite/allwpilib #2821: PIDController atSetPoint Returns True Until Calculate

JIceberg, 2020-10-31:
For preface: I'm looking at the PIDController in wpilibj. I haven't looked yet at the one for wpilibc.

**Describe the question you have.**
I'm wondering why this is the behavior of the PIDController? Based on the unit testing, where I saw this:
```java
controller.setTolerance(kTolerance);
      controller.setSetpoint(kSetpoint);

      controller.calculate(0.0);

      assertFalse(controller.atSetpoint(),
          "Error was in tolerance when it should not have been. Error was "
          + controller.getPositionError());
```
This is an expected, and intended, behavior according to these tests, but I think this is a confusing feature.

**Describe the reason for your confusion.**
I don't see anywhere in the docs where this is noted. Below is where one would believe it would be mentioned, but all it says is "specify tolerance before calling atSetpoint".

[image]

Furthermore, it makes the use of a control loop with the PIDController somewhat of a pain. If I want to run a control loop, I have to call `controller.calculate()` before I can test the setpoint. Which means my custom control loops are going to be using do-while if I don't want to add that clunky calculation before running the loop.

**Is your question related to a problem? Please describe.**
I don't see any previous mention of this anywhere. If I'm wrong, correct me.

**Additional context**
A conversation I was having when trying to work with my implementation of WPILib's PIDController:
[image]

calcmogul, 2020-10-31:
> If I want to run a control loop, I have to call controller.calculate() before I can test the setpoint. Which means my custom control loops are going to be using do-while if I don't want to add that clunky calculation before running the loop.

It's very unusual that you're using a while loop.

JIceberg, 2020-10-31:
I'm referring to an implementation of the controller in a simple loop--separate from something periodic. For example, if I am just using the controller as is and do
```java
PIDController controller = new PIDController(kP, kI, kD); // arbitrary values
controller.setSetpoint(10);
while (!controller.atSetpoint()) {
  // perform some control loop
  // calculate() is called at some point
}
```
it wouldn't work with the way `atSetpoint()` is currently handled (it completely bypasses the loop). I guess some additional context is that I was just trying to test the controller's behavior. The following does work however,
```java
PIDController controller = new PIDController(kP, kI, kD); // arbitrary values
controller.setSetpoint(10);
do {
  // perform some control loop
  // calculate() is called at some point
} while (!controller.atSetpoint());
```
I can see why this is an unusual implementation, though.

PR #2822 (2020-11-28): Recompute the error in AtSetpoint()
This makes AtSetpoint() return false after the setpoint is changed with
SetSetpoint().

Closes #2821.
commit 3413bfc0 (2020-11-28): [wpilib] PIDController: Recompute the error in AtSetpoint() (#2822)

This makes AtSetpoint() return false after the setpoint is changed with
SetSetpoint().

Closes #2821.
closed 2020-11-28 by PeterJohnson