wpilibsuite/allwpilib #2836: [simulation] macOS: Exception thrown when simgui window is closed

prateekma, 2020-11-06:
```
java(35247,0x700000987000) malloc: *** error for object 0x7ff79a4639c0: pointer being freed was not allocated
java(35247,0x700000987000) malloc: *** set a breakpoint in malloc_error_break to debug
./build/gradlerio_simulateJava.sh: line 6: 35247 Abort trap: 6           "/Users/prateekma/wpilib/2021/jdk/bin/java" -Djava.library.path="/Users/prateekma/Documents/Testing/ramsetedrivetrainsim/build/tmp/jniExtractDir" -XstartOnFirstThread -jar "/Users/prateekma/Documents/Testing/ramsetedrivetrainsim/build/libs/ramsetedrivetrainsim.jar"
```

Happens when the GUI application is closed by clicking on the red X in the bottom left corner and also happens on Ctrl+C in the terminal.

PeterJohnson, 2020-11-07:
Is this with any vendor deps?  There's really not enough information to debug here.

prateekma, 2020-11-07:
Here is the full exception: https://hastebin.com/ekazatezej.yaml
There are no vendor dependencies.

closed 2020-12-04 by PeterJohnson