wpilibsuite/allwpilib #2842: Simulation XBox Axis incorrect

agasser, 2020-11-10:
**Describe the bug**
XBox controller axes do not have the correct index values in the simulator. The values differ from what the Driver Station displays and when the XBoxController class returns.

| Axis | DS Index | Sim Index|
| ------ | ------ | ------ |
| L X | 0 | 0 |
| L Y | 1 | 1 |
| L Trigger | 2 | 4 |
| R Trigger | 3 | 5 |
| R X | 4 | 2 |
| R Y | 5 | 3 |

**To Reproduce**
1. Launch the simulator
1. Drag a XBox Controller from _System Joysticks_ to _Joysticks_
1. Launch the Driver Station
1. Operate the axes and observe the difference between Driver Station and the simulator

**Expected behavior**
Driver Station and the simulator should report the same axes values.

**Screenshots**
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Java version: 11 (packaged with WPILib)

ThadHouse, 2020-11-10:
You need to use the Map Gamepad checkbox to force the controllers to correctly map.

agasser, 2020-11-10:
Thanks, that did fix the problem.
Is that documented anyplace? It seems like the box should be checked by default, why would an FRC team want the box unchecked. Code written to the unchecked behavior would not function properly with a real robot using Driver Station.