wpilibsuite/allwpilib #2869: pauseTiming() fills sim values with garbage

Starlight220, 2020-11-16:
If `SimHooks.pauseTiming()` is used from code, all `SimValue`s are 
filled with garbage such as `NaN`, infinities, or other very large negative values. 
Pausing from the SimGUI doesn't have this bug, only when pausing from code.

PeterJohnson, 2020-11-16:
Can you post an example test that demonstrates this?

Starlight220, 2020-11-16:
[This branch](https://github.com/Starlight220/WPILib2021Beta/tree/minimal) is the most minimal example I currently have.

I call `SimHooks.pauseTiming()` in [`autonomousInit()`](https://github.com/Starlight220/WPILib2021Beta/blob/0f3c8ea6d2269612669299d23e8821c346cd3ba6/src/main/java/frc/robot/Robot.java#L52), and on transition to auto the timing is stopped (as it should). However, when resuming (by either `Resume` or `Step`), values change to garbage.

PeterJohnson, 2020-11-16:
I’m not sure if this is causing your issue, but you really don’t want to be calling things like pauseTiming() from within your robot program.  Those functions are only designed to be used from test code.

Starlight220, 2020-11-16:
Yeah, though the same thing happens when called from test code (see the test files).

PeterJohnson, 2020-11-17:
It looks like what's happening here is the NavX implementation doesn't like timing being paused, which causes its yaw output to be nan.  That nan then propagates through your code.

Starlight220, 2020-11-17:
The encoders also get filled with `inf` etc. I doubt that the NavX `NaN` pollutes the raw encoder values too. 
Also, how is `pauseTiming()` any different from the `Pause` button in the SimGUI? Only the former causes this problem.

PeterJohnson, 2020-11-17:
There is zero difference between calling pauseTiming() and the Pause button in the GUI.  They call the exact same HALSIM_PauseTiming() function.

Your code simulationPeriodic() function sets the encoder distance and rate from the drivetrain simulator state, which is fed from the gyro... so the NaNs are propagating through your code.

PeterJohnson, 2020-11-17:
It appears the issue is due to initializing the NavX while timing is paused.  If you take out the pauseTiming() from your code, and run the GUI by *first* pausing timing, then starting auto, then stepping timing, you'll see the same symptom.

Starlight220, 2020-11-17:
Is there a way to make sure that the NavX is initialized before pausing?

PeterJohnson, 2020-11-17:
Oh, whoops, it maybe isn't the NavX.  You're setting m_gyroAngleSim, so that could be coming from your code too.  I would recommend stepping through your code to see where the NaN's start from.

Starlight220, 2020-11-17:
The root cause of this bug was found, reported in #2870

closed 2020-11-17 by Starlight220