wpilibsuite/allwpilib #2872: JoystickSim does not always work as expected

CoolSpy3, 2020-11-19:
**Describe the bug**
When using the `JoystickSim` class, a number of problems prevent you from using it to its fullest ability. Firstly, after calling, `setX`, `setY`, etc, you are supposed to call `notifyNewData` to "commit" the changes. I found that I had to call this twice rather than once to achieve the desired effect (consistently). Additionally, when using `JoystickSim` in a `Command`, I found that I had to create a new version every cycle. Despite these changes, `JoystickSim` still occasionally (once every ~10 cycles, so ~5 times / second) set the joystick axis to 0 instead of the desired value.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a `Joystick`
2. Create a `JoystickSim` using it
3. Repeatedly create a new value
4. Use it to call `setX` on `JoystickSim`
5. Call `notifyNewData`
6. Print the joystick's x and your intended x.
7. Try the same code as part of a `Command`

**Expected behavior**
The above tests should always produce identical values for the desired and actual values.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Java version 11
 - C++ version N/A

**Additional context**
This is the RobotContainer of an otherwise blank test project which can be used to produce the errors. It already has the aforementioned fixes applied.
[RobotContainer.zip](https://github.com/wpilibsuite/allwpilib/files/5564624/RobotContainer.zip)

PeterJohnson, 2020-11-19:
First of all, are you running this with the simulation GUI enabled or the DS connected?  You need to make sure to disable the DS functionality of the GUI if you're setting joystick data yourself.  That is most likely why you're seeing intermittent 0s and needing to create JoystickSim more than once, because the GUI is also calling notifyNewData() and setting joystick data.  Of course disabling the DS functionality also means you will have to enable and disable the robot through code as well--if you want to test this type of functionality, it's better done in a unit test.

CoolSpy3, 2020-11-19:
Thank you, that seems to have fixed the problem. However, our team would still be interested in running the simulator with halsim_gui. Is there a way to override joystick outputs while the gui is running?

PeterJohnson, 2020-11-19:
You can disable the DS in the GUI via Options | Turn off DS.  But note doing so disables the "Robot State" control as well... Robot State and joystick inputs are intrinsically linked throughout the HAL and DriverStation classes right now so this is not an easy thing to separate.

CoolSpy3, 2020-11-19:
Okay, we'll look into it. Thanks for your help.

closed 2020-11-19 by CoolSpy3