wpilibsuite/allwpilib #2890: Sim GUI device names should be set programmatically

pjreiniger, 2020-11-26:
**Is your feature request related to a problem? Please describe.**
Currently the "display names" for sim devices are set in the hal sim gui, and saved in an INI file which is included in the default gitignore. That means team members will potentially have different looking displays on their computer, or duplicate the work of setting everything up. In addition, this is fragile for refactoring; swapping port numbers for two devices will result in the wrong name being displayed.

**Describe the solution you'd like**
Add a function in the devices themselves that can go down to the HAL, or add it to the sim wrappers (i.e add a `setDisplayName` and `getDisplayName` to either the `Encoder` class, or the `EncoderSim` class). This will tie the name directly to the device to avoid any refactoring headaches, and it can be persisted through any simulation platform (hal gui, websocket, fancy unity things, etc)

I'd be happy to take a crack at it, just would like the opinion on if it should be in the real class or the sim class.

PeterJohnson, 2020-11-26:
It should definitely be in the real class.

There is a setName() in the Sendable interface for use with LiveWindow, but we've deprecated that in favor of SendableRegistry.setName().  That function is used by RobotBuilder to set the display name in LiveWindow / dashboards.  It would be kind of nice to only have a single function for setting a name, but setName() has a ton of overloads so it would be kind of painful to undeprecate and overload it.

If done as a separate HAL interface for each device type, note this will require a fair amount of grunt work (HAL API, HALSIM API, HAL and HALSIM JNI wrapper functions, WPILibC/J functions, etc).

pjreiniger, 2020-11-29:
It has to make its way down to the HALSIM api regardless in order for the GUI to use it, so all those API's need to be touched anyway.

I went down the later path in #2894 , since it sounds much more invasive to un-deprecate the functions, and since the grunt work had to be done anyway