wpilibsuite/allwpilib #2907: Annoyance: SendableChooser does not create Selected entry

prensing, 2020-12-04:
**Is your feature request related to a problem? Please describe.**
On creation (in Java at least), SendableChooser does not create the NT entry for "/selected". This only gets created when a dashboard actually makes a selection. This is annoying because the simpler tools which don't know/use the Chooser structure cannot change the value. For instance, the NT browser in the new Simulation GUI and OutlineViewer can be used to change a Chooser by typing into /selected, but only after a Dashboard is used first.

**Describe the solution you'd like**
The creation code should make the /selected entry in NT so that the data is fully populated.

**Describe alternatives you've considered**

**Additional context**
Java. I have not test C++. 
And, yes, this is a small feature, not any kind of serious problem.

Starlight220, 2020-12-10:
I'll follow up with a relevant question: what's the difference between `active` and `selected`?

sciencewhiz, 2021-01-03:
I believe this is intentional. If the robot created the entry, the dashboard's value wouldn't "win" on a reconnect.

prensing, 2021-01-03:
Well, I hear you, but is it really any different than a floating point number? Those are always created by the robot and the Dashboard wins if the robot restarts.

auscompgeek, 2021-01-03:
The lack of a selected entry also indicates the lack of a selection as well, which you would not be able to represent on NetworkTables if the SendableChooser does not have a default option.

prensing, 2021-01-03:
But if it does have a default option (which we have always done), why not preload that as the selected entry?

prateekma, 2021-01-03:
Going back to the original issue,

> This only gets created when a dashboard actually makes a selection. This is annoying because the simpler tools which don't know/use the Chooser structure cannot change the value.

The simulation GUI now has a proper widget for a SendableChooser (under the NetworkTables menu option).

prensing, 2021-01-03:
Sounds good, but I am not seeing it. Is that only in the 2021 Beta version? I think I am currently using the dev version from 2020.

prateekma, 2021-01-03:
Oh yes, it was introduced in 2021.1.1-beta-3.

PeterJohnson, 2021-03-15:
Closing as a won't fix.  As stated above, this is intentional so that the dashboard will always win on reconnect.  If the robot set the value, this might not happen and you can end up with the robot value overwriting the dashboard value, leading to e.g. the wrong auto being selected, and a single value for both bit a number of teams in years before this was split.  The split of the two keys was made very intentionally to make SendableChooser reliable for the purpose teams use it for.

closed 2021-03-15 by PeterJohnson