wpilibsuite/allwpilib #2921: Simulated Noise makes Glass Display Hard to read

AustinShalit, 2020-12-07:
The simulated noise introduced in #2903 makes the data displays hard to read because the value keeps getting redrawn.

AustinShalit, 2020-12-08:
For context: https://youtu.be/pFE1MRhsyLE

Starlight220, 2020-12-08:
This happens in everything that displays live data in numerical format - all dashboards have this problem with quickly changing values. The sim noise causes the values to change even more than usual. 
Is it harder to read the values in Glass than in Shuffleboard or any of the other dashboards? If not, I don't think much can be done about this, and it isn't a new issue.

TheTripleV, 2020-12-09:
I think it would make it easier on the eyes if space for the "-" was reserved so values flipping signs don't shift.

closed 2026-11-03 by AustinShalit