wpilibsuite/allwpilib #2933: Make UnscentedKalmanFilter use square-root formulation

calcmogul, 2020-12-12:
A major source of computational overhead in the Unscented Kalman filter is the cholesky decomposition used in the sigma point calculations. By using the square root of the covariance matrices and using the rank-1 Cholesky update/downdate, we can save a lot of computational work.

Here's a paper on it:
https://www.researchgate.net/publication/3908304_The_Square-Root_Unscented_Kalman_Filter_for_State_and_Parameter-Estimation

At least based on that paper's results, it can be even more efficient than the Extended Kalman filter (I imagine that depends on the model). Since the UKF is more accurate, there'd be essentially no reason to use an EKF.

connorworley, 2022-04-11:
I'd be interested in taking a crack at this if nobody is currently working on it. I could use some help validating my implementation against the math once I have something.

calcmogul, 2022-04-11:
I haven't started it yet, so go ahead. Eigen has an implementation of the Cholesky rank 1 up/downdate that may be helpful (see https://github.com/sfwa/ukf/blob/master/include/UKF/Core.h#L297-L303 for a usage example). Start in C++ first, then we'll figure out the JNI parts later.

connorworley, 2022-04-13:
Hey Tyler, do you have any literature I can look at explaining the discretiziation steps in https://github.com/wpilibsuite/allwpilib/blob/765efa325e07dc18a69bf6f6daa8d036da1b922d/wpimath/src/main/native/include/frc/estimator/UnscentedKalmanFilter.h#L230-L243?

The reason I ask is that Eqns. 20 and 24 in Merwe 2001 involve the square root process (Q) and measurement (R) noise covariance matrices. We can discretize sqrt(R) by passing sqrt(dt) to DiscretizeR instead of dt, but I'm less sure this works for sqrt(Q) and DiscretizeAQTaylor. Any pointers on this part? I think I've got the rest more or less figured out.

calcmogul, 2022-04-13:
[image]
This might be a hint: https://math.stackexchange.com/questions/3395839/square-root-and-exponential-of-diagonalized-matrix

I don't know enough linalg to figure this out. If we can't, our options are either use an approximate discretization or do it the long way by using Q.sqrt() from "unsupported/Eigen/MatrixFunctions".

connorworley, 2022-04-14:
I've got a draft implementation up at https://github.com/wpilibsuite/allwpilib/pull/4168. Still have some open questions about the API that I'd like to discuss there.

closed 2022-06-09 by PeterJohnson

connorworley, 2022-06-09:
Should probably have a followup issue for the discretization optimizations discussed here and in the PR.