wpilibsuite/allwpilib #2936: Simulating a Turn in Place Oscillates Indefinitely

jonathandao0, 2020-12-14:
**Describe the bug**
Attempting to implement [TurnToAngleProfiled.java](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/gyrodrivecommands/commands/TurnToAngleProfiled.java) within the [statespacedifferentialdrivesimulation](https://github.com/wpilibsuite/allwpilib/tree/master/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/statespacedifferentialdrivesimulation) example will cause the robot to oscillate indefinitely. I've tried adjusting this to a regular PIDCommand, as well as creating a regular Command that implements the controllers with no real success. Adjusting the PID gains to extreme values seems to make things worse (e.g. Reducing the P value to 0.000007 makes the robot spin indefinately). The only thing that seems to work is clamping the output of the PIDController to +-0.1, but increasing it to +/-0.2 still has the robot spinning indefinitely.

Also, there is no way to get the output of the ProfiledPIDController/PIDController without calling calculate(), which made it hard to validate if the PID math was correct.

prateekma, 2020-12-14:
It essentially looks like the robot is turning the wrong way to attempt to correct the error. As you described, when the TurnToAngleProfiled command is copied into the StateSpaceDifferentialDrive example and ran, if you plot the current angle vs. the desired angle, you can see that the two values actually diverge when the command is scheduled:

[image]

This happens because the `arcadeDrive` method attempts to rotate the robot clockwise for positive angle error; however, this is the opposite of what it should be doing. Simply adding a negative sign before `output` on Line 38 of `TurnToAngleProfiled.java` corrects this behavior. Note that the gains in the example are just placeholder gains and the default P value of 1.0 causes severe oscillation around the setpoint. However, I was able to get acceptable results after bringing this value down to 0.2.

We will look into fixing the `GyroDriveCommands` example.

jonathandao0, 2020-12-15:
Whoops. I rechecked my code and realized the portion that was supposed to pull PID values from SmartDashboard wasn't actually pulling the values. Now I feel a bit silly opening this in the first place. Thanks!

closed 2020-12-15 by jonathandao0