wpilibsuite/allwpilib #2948: Make NT bind() to check for access failure

Starlight220, 2020-12-20:
```
NT: ERROR: bind() to port 1735 failed: An attempt was made to access a socket in a way forbidden by its access permissions. (TCPAccepter.cpp:118)
```
>Most likely that NT error is due to an aggressive firewall.  We bind the NT socket globally, so eg it’s accessible from the network, not just localhost
> It seems like the OS in that case is blocking the entire bind() rather than just restricting it to localhost

Starlight220, 2022-10-30:
Is this related/duplicate of #1650?

PeterJohnson, 2022-10-31:
No, because NT3 didn’t use libuv.  However, this can be closed because the code has been completely rewritten and I’ve never been able to reproduce this.