wpilibsuite/allwpilib #2949: Feature Request: Simulate RIO File Deploy System

CoryNessCTR, 2020-12-22:
When simulating robot code, the files inside the deploy directory are not put anywhere. This can be an issue for things such as config files that you want to verify parse correctly when simulating.

I'd like some form of file system that's simulated. For example, a new directory that's created that mimics the RIO file system, with a `/home/lvuser/deploy` path that files inside the deploy directory are copied to.

prateekma, 2020-12-22:
You can call `Filesystem.getDeployDirectory()` to get the location of the deployed files on both the roboRIO and in Desktop Sim.

AustinShalit, 2020-12-22:
Here is the method documentation and implementation: https://github.com/wpilibsuite/allwpilib/blob/21d949daa8d4546bedf06f78d36526708de84bee/wpilibj/src/main/java/edu/wpi/first/wpilibj/Filesystem.java#L47-L62

CoryNessCTR, 2020-12-22:
Yep, that achieves what I want.
Thanks guys!

closed 2020-12-22 by CoryNessCTR