wpilibsuite/allwpilib #2950: Logitech X-mode controller Raw Axes in simulation do not match physical robot

bhall-ctre, 2020-12-22:
**Describe the bug**
Using a Logitech controller in X-mode, raw axes 2-5 are different in simulation than on a physical robot (using the driver station).

On a robot:
- Raw Axis 2 is the left trigger, value starts at 0 and goes up to 1
- Raw Axis 3 is the right trigger, value starts at 0 and goes up to 1
- Raw Axis 4 is the right joystick's horizontal direction
- Raw Axis 5 is the right joystick's vertical direction

In simulation:
- Raw Axis 2 is the right joystick's horizontal direction
- Raw Axis 3 is the right joystick's vertical direction
- Raw Axis 4 is the left trigger, value starts at -1 and goes up to 1
- Raw Axis 5 is the right trigger, value starts at -1 and goes up to 1

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in a Logitech controller in X-mode
2. Open the FRC Driver Station
3. Run a robot program in simulation with the GUI

**Expected behavior**
The Logitech controller acts identically in the driver station and in simulation.

**Screenshots**
No axes pressed: 
[image]

Right joystick forward (note that Axis 5 changed in DS but Axis 3 changed in sim):
[image]

Left trigger pressed (note that the left trigger went from 0 to 1 in DS (Axis 2) but -1 to 1 in sim (Axis 4)):
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 x64
 - Java version 11.0.8
 - C++ version 17

**Additional context**
Logitech X-mode is supposed to act like an Xbox controller, so this may also be an issue with Xbox controllers in general.
This issue also affects joystick.GetRawAxis() calls in code.

prateekma, 2020-12-22:
Can you try selecting the "Map Gamepad" option to see if it fixes th axes?

bhall-ctre, 2020-12-22:
Yes, "Map Gamepad" does fix the issue. That should probably be checked by default so gamepads in simulation match the driver station out of the box.

ThadHouse, 2020-12-22:
Making it default breaks other joysticks. Like the Attack 3 joysticks are wrong if that is selected. There are slight differences how the DS reads joysticks. So picking a default will always break someone.

PeterJohnson, 2020-12-22:
There's an open issue to try to better guess a default here, but we're always going to need a checkbox as we can't implement the same detection the DS does, particularly cross platform.

Another workaround on Windows is to use the halsim_ds_socket simulation extension to get data from the real DS instead of using the simulation GUI DS.

bhall-ctre, 2020-12-22:
Yeah, that makes sense. Thanks!

closed 2020-12-22 by bhall-ctre