wpilibsuite/allwpilib #2970: Deduplicate angle normalization functions

calcmogul, 2020-12-26:
We have `frc::NormalizeAngle()` in AngleStatistics.h, `units::math::NormalizeAngle()` in units/math.h, and `frc::GetModulusError()` in frc/controller/ControllerUtil.h. They all do the same thing, but with varying implementations and argument types.

|Name|Location|Argument type|Branchless?|
|----|--------|-------------|-----------|
|`frc::NormalizeAngle()`|`wpilibc/src/main/native/include/frc/estimator/AngleStatistics.h`|double|no|
|`units::math::NormalizeAngle()`|`wpimath/src/main/native/include/units/math.h`|units::radian_t|yes|
|`frc::GetModulusError()`|`wpilibc/src/main/native/include/frc/controller/ControllerUtil.h`|arbitrary (template)|yes|

`frc::GetModulusError()` is also more general in that it can handle asymmetric input boundaries; the other two are hardcoded to -pi to pi.

Starlight220, 2020-12-28:
So I assume we want to keep only `frc::GetModulusError()`?

calcmogul, 2020-12-28:
Yea. We can generalize it as something like
```cpp
// Copyright (c) FIRST and other WPILib contributors.
// Open Source Software; you can modify and/or share it under the terms of
// the WPILib BSD license file in the root directory of this project.

#pragma once

namespace frc {

/**
 * Returns modulus of input.
 *
 * @param input        Input value to wrap.
 * @param minimumInput The minimum value expected from the input.
 * @param maximumInput The maximum value expected from the input.
 */
template <typename T>
constexpr T InputModulus(T input, T minimumInput, T maximumInput) {
  T modulus = maximumInput - minimumInput;

  // Wrap input above maximum input
  int numMax = (input + maximumInput) / modulus;
  input -= numMax * modulus;

  // Wrap input below minimum input
  int numMin = (input + minimumInput) / modulus;
  input -= numMin * modulus;

  return input;
}

}  // namespace frc
```

Where the input would be the error (reference - measurement) for current uses of GetModulusError(). I'm not sold on the name `InputModulus` though. Maybe we should also provide a convenience function that does

```cpp
template <typename T>
constexpr T AngleModulus(T angle) {
  return InputModulus<T>(angle, T{-wpi::math::pi / 2.0}, T{wpi::math::pi / 2.0});
}
```

PR #2998 (2021-01-02): Deduplicate angle modulus functions
frc::NormalizeAngle(), units::math::NormalizeAngle(), and
frc::GetModulusError() were replaced with frc::InputModulus() and
frc::AngleModulus().

They were placed in wpimath/src/main/native/include/frc/MathUtil.h for
C++ and wpimath/src/main/java/edu/wpi/first/wpiutil/math/MathUtil.java
for Java. Feedback is welcome on where the best place is to put these
functions.

Closes #2970.
closed 2021-01-02 by PeterJohnson