wpilibsuite/allwpilib #3002: LiveWindow Drive Object motor inversion causes jittering

AustinShalit, 2020-12-30:
**Describe the bug**
An inverted motor connected to a Drive object when controlled via LiveWindow will jitter between positive and negative values. This happens on both real robots and in simulation.

**To Reproduce**
Steps to reproduce the behavior:
1. `./gradlew :wpilibjExamples:runmecanumdrive`
2. Enable test mode
3. Set one of the left motors to any value

**Expected behavior**
Motors are controlled properly.

**Screenshots**

https://user-images.githubusercontent.com/3799260/103322985-87daa300-49f5-11eb-8325-abde8f3ef4f8.mov

Starlight220, 2020-12-30:
Is the problem with the mecanum object, the motor objects, LW, or something else entirely?

AustinShalit, 2020-12-30:
> Is the problem with the mecanum object, the motor objects, LW, or something else entirely?

No clue. My guess is that the motor's LW code and drive object's LW code are fighting.

AustinShalit, 2020-12-30:
Here are two potential solutions:
```
diff --git a/wpilibj/src/main/java/edu/wpi/first/wpilibj/drive/MecanumDrive.java b/wpilibj/src/main/java/edu/wpi/first/wpilibj/drive/MecanumDrive.java
index 8189e68a5..d87c551fc 100644
--- a/wpilibj/src/main/java/edu/wpi/first/wpilibj/drive/MecanumDrive.java
+++ b/wpilibj/src/main/java/edu/wpi/first/wpilibj/drive/MecanumDrive.java
@@ -258,7 +258,9 @@ public class MecanumDrive extends RobotDriveBase implements Sendable, AutoClosea
     builder.setActuator(true);
     builder.setSafeState(this::stopMotor);
     builder.addDoubleProperty(
-        "Front Left Motor Speed", m_frontLeftMotor::get, m_frontLeftMotor::set);
+        "Front Left Motor Speed",
+        () -> m_frontLeftMotor.get() * (m_frontLeftMotor.getInverted() ? -1 : 1),
+        m_frontLeftMotor::set);
     builder.addDoubleProperty(
         "Front Right Motor Speed",
         () -> m_frontRightMotor.get() * m_rightSideInvertMultiplier,
```

```
diff --git a/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWMSpeedController.java b/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWMSpeedController.java
index ea4b7aaf8..6dfa72a8b 100644
--- a/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWMSpeedController.java
+++ b/wpilibj/src/main/java/edu/wpi/first/wpilibj/PWMSpeedController.java
@@ -46,7 +46,7 @@ public abstract class PWMSpeedController extends PWM implements SpeedController
    */
   @Override
   public double get() {
-    return getSpeed();
+    return getSpeed() * (m_isInverted ? -1.0 : 1.0);
   }
   @Override
```

PR #3016 (2020-12-31): [wpilib] PWMSpeedController.get(): Apply Inversion
Closes #3002
closed 2020-12-31 by PeterJohnson