wpilibsuite/allwpilib #3041: Elevator Sim Example gearing doesn't work for large values

sciencewhiz, 2021-01-03:
**Describe the bug**
If the Elevator Gearing in the Elevator Sim example is increased above 40 or so, the behavior does not change as expected. The elevator ramps up and down almost instantaneously. With values between 5 & 30, it does work as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new Elevator Sim Example project
2. Change the Elevator Gearing from 10.0 to 100.0
3. Change the output when the trigger is hit to set(1.0) instead of the PID value, for minimizing variables

**Expected behavior**
The elevator encoder value should slowly ramp up, 10x slower then the default gearing of 10. It should also more slowly ramp down.

**Screenshots**
10x:

https://user-images.githubusercontent.com/4885091/103488238-c4c0e400-4dbf-11eb-8974-985b27329af0.mp4

100x:

https://user-images.githubusercontent.com/4885091/103488232-becb0300-4dbf-11eb-891f-73c573cb1c20.mp4

mcm001, 2021-01-04:
[image]
After digging into the code, it seems that Runge Kutta is giving wildly different results than the regular `plant.calculateX` method.

mcm001, 2021-01-04:
[image]
100:1 

10:1

calcmogul, 2021-01-04:
That means our numerical integration with Runge-Kutta isn't accurate enough. We might have to use an integration method with adaptive step size.

mcm001, 2021-01-04:
I implemented RKF45 and it seems to be much better behaved 
[image]

mcm001, 2021-01-04:
At 100:1, a unit test over 2 seconds produces the following:
```
rkf45 final height: 0.777444366393044 meters
rk4 final height: NaN
true height from exp(): 0.8346119600301571 meters
```

PR #3047 (2021-01-07): Add RKF45 integration
This is more stable than Runge-Kutta for systems with large elements in their A or B matrices. Closes #3041
closed 2021-01-07 by PeterJohnson