wpilibsuite/allwpilib #3063: Documentation of Device Simulation

kinahawi, 2021-01-08:
The ability to simulate devices using the GUI and WebSockets is a very useful feature. To make it more accessible for the FRC population, I recommend that documentation be developed describing how to properly use the capability. This should include a guidebook (as done with many other features) and Javadoc style documentation of the relevant APIs. It would also be useful to provide an example that can be extended or tailored by users.

Starlight220, 2021-01-09:
This can be closed because of duplicates. 
The API docs part will be covered by #2714. The need for documentation on frc-docs is noted by https://github.com/wpilibsuite/frc-docs/issues/704.

Daltz333, 2021-01-09:
Nah, this issue is about documenting webhooks API. Which is a developer focused feature.

kinahawi, 2021-01-09:
For what it’s worth, I wrote this issue at the recommendation of Peter Johnson based on an exchange on a Chief Delphi thread.