wpilibsuite/allwpilib #3068: [actions] Create an action that comments on PRs when code affects documentation

Daltz333, 2021-01-09:
Any code that has been referenced or included in [frc-docs](https://github.com/wpilibsuite/frc-docs) should have an associated documentation PR. We should create an action or CI that will

1. Grabs the current diff
2. Create a list of URLs that will then be matched with frc-docs source
3. Comment on the PR that the associated frc-docs line # has been affected and to update that line.

Starlight220, 2022-05-02:
@Daltz333 isn't this solved by Inspector?
Or could be fixed by one of #3069 / #3672?

I'm honestly not sure if it'll be that significant of a workflow improvement compared to the current setup with Inspector.

Daltz333, 2022-05-02:
Inspector does not automatically update when code changes. This would be useful and push the burden to whoever updates the example, as it should be IMO.