wpilibsuite/allwpilib #3070: WPILib requests the wrong URL from the Maven repo while running in an ARM Docker container

khari05, 2021-01-09:
**Describe the bug**
WPILib requests an OpenCV image from the FRC Maven repo that doesn't exist when running inside of an ARM Docker container.

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a machine with Docker for Apple Silicon
2. Clone [my repo](https://github.com/khari05/FRC-test-docker) to the computer
3. Open the root folder of the repo in a terminal
4. Run `docker-compose run java bash`
5. Run `./gradlew build`

**Expected behavior**
It should correctly download the dependencies then successfully compile.
**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Docker for Mac 
 - Docker Base Image: openjdk:11
 - Java version: OpenJDK 11.0.9.1
 - C++ not installed

**Additional context**
It seems to be looking for an OpenCV file that doesn't exist at `frcmaven.wpi.edu` and then fails. The program does compile correctly on Docker for Mac on Intel CPUs and in GitHub Codespaces. I haven't been able to test the Docker container on any other ARM devices.

PeterJohnson, 2021-01-09:
We don't build 32-bit linux binaries.  Do you have the option of running a 64-bit docker container?

PeterJohnson, 2021-01-09:
Can you try the wpilib/roborio-cross-ubuntu image instead?

khari05, 2021-01-09:
The OpenJDK Docker only compiles to arm64 so I should be running a 64 bit docker container.

ThadHouse, 2021-01-09:
It's being detected as a 32 bit JDK. Since we don't ship 32 bit x86 binaries, in order for that to work, that image would need to be detected as having a 64 bit JDK. We have no plans to reintroduce 32 bit x86 binaries. What happens if you use the wpilib docker image? That is known to have everything correct, and is recommended over using other docker images.

PeterJohnson, 2021-01-09:
We've not tested any linux/arm64 containers.  Use an x86-64/amd64 container.

khari05, 2021-01-09:
Changing the docker image to `wpilib/roborio-cross-ubuntu:2021-18.04` fixes my issue. I guess WPILib didn't read OpenJDK for arm64 correctly.

closed 2021-01-09 by khari05

ThadHouse, 2021-01-09:
We don't support arm64 natively, so that would be detected as 32 bit Linux.