wpilibsuite/allwpilib #3071: Incorrect SimDevice(Sim) device names in test?

brettle, 2021-01-10:
https://github.com/wpilibsuite/allwpilib/blob/a751fa22d20e9fc268998c7c2e522226714f278b/wpilibj/src/test/java/edu/wpi/first/wpilibj/simulation/SimDeviceSimTest.java#L141-L143

Why is the name of the `SimDeviceSim` different than the name of the `SimDevice` in this test? Seems like they should be the same, or the test should not be passing. Am I missing something?

CoolSpy3, 2021-01-10:
`SimDevice` is the class which "creates" `SimDevice`s and `SimValue`s. `SimDeviceSim` provides methods which alter `SimValue`s and detect updates to `SimDevice`s or `SimValue`s.

brettle, 2021-01-10:
> `SimDevice` is the class which "creates" `SimDevice`s and `SimValue`s. `SimDeviceSim` provides methods which alter `SimValue`s and detect updates to `SimDevice`s or `SimValue`s.

You misunderstand my question. Why is "testV**C**1" passed to `SimDevice.create` but "testV**M**1" pased to `SimDeviceSim`?

CoolSpy3, 2021-01-10:
Upon inspection, it looks like this is a typo. However, it does not create an error because the value callbacks utilize the value handle and not the device handle, so any name can be passed to `SimDeviceSim`. These methods should probably be made static or moved to `SimValue`.

brettle, 2021-01-10:
I think the test is doubly wrong and the errors cancel out. The typo is one bug. The other is that the test should call `SimBoolean simVal = sim.getBoolean("v1");` and pass `simVal` instead of `val` to `sim.registerValueChangedCallback()`. I don't recommend making anything static since doing so would break compatibility. It would be better if `registerValueChangedCallback()` complained if the `Sim*` it received was not associated with the `SimDeviceSim`.

Starlight220, 2021-01-10:
> Upon inspection, it looks like this is a typo. However, it does not create an error because the value callbacks utilize the value handle and not the device handle, so any name can be passed to `SimDeviceSim`. These methods should probably be made static or moved to `SimValue`.

This makes zero sense. `SimDeviceSim` gets a native handle by matching the name. If the names don't match, how does it get the right native handle?

CoolSpy3, 2021-01-10:
> This makes zero sense. `SimDeviceSim` gets a native handle by matching the name. If the names don't match, how does it get the right native handle?

It's able to do this because the value callback methods utilize the handle from the passed `SimValue`, not the device's native handle.

PR #3095 (2021-01-17): Update SimDeviceSimTest
Fixes #3071.
closed 2021-01-17 by PeterJohnson