wpilibsuite/allwpilib #3074: How to find standard deviation values for DifferentialDrivePoseEstimator

LudumDareDevelopment, 2021-01-10:
Hi,

I just read about the new DifferentialDrivePoseEstimator, and I am wondering if there is a methodical way we can come up with the standard deviations for the computed states vs measurements. We probably won’t be using vision/PnP for correcting the observer, but I do want to know if there is a way to find out (statistically or by plotting charts) whether we should trust the drivetrain model more or our encoder measurements more. I was thinking something like drive the robot straight forward a known distance, and then record the state estimate vs encoder/gyro measurements at the end, and repeat this a few times. Then do some statistical analysis. But I’m not sure I’m going about this the right way. Based on what I understand the state estimator is integrating the wheel and angular velocities (which might be inaccurate in practice with NEO inbuilt encoders which have low CPR) and correcting with the left meter/right meter position measurements (I would expect these to be more accurate because positional data on lower CPR encoders is generally pretty good). So, at least for NEO motor robots, I think we would probably want to set the measurement standard devs lower, but I just don’t know how much or how to tune correctly to get the most accurate pose. Any help would be greatly appreciated.
Thanks in advance,
LudumDareDevelopment

prateekma, 2021-01-10:
These are great questions and we will be working on adding some of my answers below into the documentation over at docs.wpilib.org as well.

Given that you won't be using Vision/PnP measurements (what we refer to as "global" measurements), I will just go over the standard deviations for the state and the "local" measurements (encoder distances and gyro angle).

Your thoughts about driving the robot, recording the measurements from your sensors, and doing some statistical analysis on those measurements is a great idea. In addition to this, you can "record" the inputs to the system (velocities) and the measurements (distances and angles) and then "replay" them in simulation with different standard deviations for the state and measurement to see how the estimator behaves in different scenarios. This approach is advantageous because you can do it completely offline (once you have collected the data on the real robot, of course).

I also agree with you that setting the local measurement standard deviations lower will give you more accurate results -- distance measurements from encoders tend to be more accurate than velocity measurements (velocity measurements also involve a lot of filtering, which can introduce latency). The regular ``DifferentialDriveOdometry`` class also only uses positions.

LudumDareDevelopment, 2021-01-10:
Thank you so much for your helpful answer. So I think we will probably stick to the normal DifferentialDriveOdometry using positional measurements, until we get a Pnp going.

LudumDareDevelopment, 2021-01-10:
One more question I had: Is there any advantage with using the state estimator and fusing it with the positional data?

prateekma, 2021-01-10:
No, not really. Where the pose estimator is primarily advantageous is if you are fusing it with vision measurements.

LudumDareDevelopment, 2021-01-10:
Got it, thanks!

closed 2021-01-10 by LudumDareDevelopment