wpilibsuite/allwpilib #3077: Description of model returned by identifyDrivetrainSystem

LudumDareDevelopment, 2021-01-11:
Hi,

I am attempting to implement a state-space controller for the drivetrain, and to do this, first I created the plant using the `LinearSystemId.identifyDrivetrainSystem` method. I couldn't find much documentation on what the states, inputs, and outputs are, but knowing that it is a N2xN2xN2 plant my best guess would be that the state is [left_velocity, right_velocity], input is [left_voltage, right_voltage], and output is [left_position, right_position], but I am not too sure. Could someone please clarify this? Thank you so much.

vmalepati1, 2021-01-11:
One more question: If I am using the `LinearSystemLoop` in a custom implementation of the `RamseteCommand` (essentially the exact same, but using the loop object to generate the voltages from the trajectory velocities, instead of using the standard feedforward+velocity PID), is it better to pass in the actual dt (the one calculated from `curTime - m_prevTime`) or the nominal dt of 0.02 s to the linear system loop's `predict` method?

LudumDareDevelopment, 2021-01-11:
One more question: If I am using the LinearSystemLoop in a custom implementation of the RamseteCommand (essentially the exact same, but using the loop object to generate the voltages from the trajectory velocities, instead of using the standard feedforward+velocity PID), is it better to pass in the actual dt (the one calculated from curTime - m_prevTime) or the nominal dt of 0.02 s to the linear system loop's predict method?

calcmogul, 2021-01-11:
The drivetrain system states are left and right velocity, inputs are left and right voltage, and outputs are left and right velocity. Use the measured dt for your predict step.

LudumDareDevelopment, 2021-01-11:
And to account for the 19.5 ms delay in velocity measurements on the Spark Max, I could use the latencyCompensate method like the following: `controller.latencyCompensate(plant, 0.02, 0.0195)`, correct?

prateekma, 2021-01-11:
That's correct.

LudumDareDevelopment, 2021-01-11:
Thank you both for your tremendous help!

prateekma, 2021-01-11:
Reopening until Javadocs/Doxygen are also updated with this answer.

PR #3080 (2021-01-12): [wpimath] Add model description to LinearSystemId Javadocs
C++ already has the model descriptions.
Closes #3077
closed 2021-01-12 by PeterJohnson