wpilibsuite/allwpilib #3093: Glass Field image pixel dimensions should be modifiable in the gui

sciencewhiz, 2021-01-17:
**Is your feature request related to a problem? Please describe.**
If you use a custom image for Field2d where the image shows more then the field, it isn't possible to set what pixels correspond to the field in the gui.

**Describe the solution you'd like**
Add gui entries for the top, left, bottom, and right entries in the imgui.ini file

**Describe alternatives you've considered**
Documenting that imgui.ini can be modified to add the field dimensions in image pixels

**Additional context**
Here's what it looks like when you import an image without modifying anything
[image]

PeterJohnson, 2021-01-17:
Note if you use a PathWeaver json file instead of a raw jpeg, the data will get loaded from there.  But it's easy to add the feature to expose it via the GUI too.

sciencewhiz, 2021-01-17:
I didn't realize you could point to a PathWeaver JSON file, since the option is to choose an image and the default filter is image file. Maybe just better documentation.

PeterJohnson, 2021-01-17:
The other problem is we don't install the PathWeaver JSON files anywhere.  You can get them on GitHub, but they really should be on the filesystem somewhere.

sciencewhiz, 2021-09-19:
Documenting the json import in wpilibsuite/frc-docs#1104 closes this.

PR #3904 (2022-01-15): Fix ProfiledPIDCommand use-after-free
Fixes #3093.
closed 2021-09-19 by sciencewhiz