wpilibsuite/allwpilib #3109: GearsBot Example SegFaults on left stick press

Modelmat, 2021-01-20:
**Describe the bug**
The GearsBot example segfaults on the left stick being pressed in teleoperated. See below for the segfault.

The Left Stick Calls the Autonomous Command:
https://github.com/wpilibsuite/allwpilib/blob/4488e25f16c6bb98cdbfa48095db56b7f6390da3/wpilibcExamples/src/main/cpp/examples/GearsBot/cpp/RobotContainer.cpp#L53

which adds the SetDistanceToBox command:

https://github.com/wpilibsuite/allwpilib/blob/4488e25f16c6bb98cdbfa48095db56b7f6390da3/wpilibcExamples/src/main/cpp/examples/GearsBot/cpp/commands/Autonomous.cpp#L27

which creates a PIDCommand with the measurement argument `m_drivetrain->GetDistanceToObstacle();`

https://github.com/wpilibsuite/allwpilib/blob/4488e25f16c6bb98cdbfa48095db56b7f6390da3/wpilibcExamples/src/main/cpp/examples/GearsBot/cpp/commands/SetDistanceToBox.cpp#L13-L16

which calls the `GetDistanceToObstacle` when run:

https://github.com/wpilibsuite/allwpilib/blob/4488e25f16c6bb98cdbfa48095db56b7f6390da3/wpilibcExamples/src/main/cpp/examples/GearsBot/cpp/subsystems/DriveTrain.cpp#L67-L70

which calls the `GetAverageVoltage()` function of AnalogInput:

https://github.com/wpilibsuite/allwpilib/blob/4488e25f16c6bb98cdbfa48095db56b7f6390da3/wpilibc/src/main/native/cpp/AnalogInput.cpp#L79-L87

which segfaults.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the GearsBot Example in C++
2. Enable teleop mode
3. Press the left stick button

**Expected behavior**
The command works? Definitely doesn't segfault

**Additional context**
```
0x00007ffff7e7b98e in frc::AnalogInput::GetAverageVoltage (
    this=0x7ffff316b752 <nt::EntryNotifier::Add(std::function<void (nt::EntryNotification const&)>, wpi::StringRef, unsigned int)+530>) at ../wpilibc/src/main/native/cpp/AnalogInput.cpp:80
80        if (StatusIsFatal()) {
(gdb) bt
#0  0x00007ffff7e7b98e in frc::AnalogInput::GetAverageVoltage() const
    (this=0x7ffff316b752 <nt::EntryNotifier::Add(std::function<void (nt::EntryNotification const&)>, wpi::StringRef, unsigned int)+530>) at ../wpilibc/src/main/native/cpp/AnalogInput.cpp:80
#1  0x00007ffff7fa8124 in std::function<double ()>::operator()() const (this=0x7fffdc015958)
    at /usr/include/c++/10.2.0/bits/std_function.h:622
#2  frc2::PIDCommand::Execute() (this=0x7fffdc0157f0)
    at ../wpilibNewCommands/src/main/native/cpp/frc2/command/PIDCommand.cpp:56
#3  0x00007ffff7fb0453 in frc2::SequentialCommandGroup::Execute() (this=0x7fffdc015dd0)
    at ../wpilibNewCommands/src/main/native/cpp/frc2/command/SequentialCommandGroup.cpp:29
#4  frc2::SequentialCommandGroup::Execute() (this=0x7fffdc015dd0)
    at ../wpilibNewCommands/src/main/native/cpp/frc2/command/SequentialCommandGroup.cpp:22
#5  0x00007ffff7f9bb82 in frc2::CommandScheduler::Run()
    (this=0x7ffff7fc77e0 <frc2::CommandScheduler::GetInstance()::scheduler>)
    at ../wpilibNewCommands/src/main/native/cpp/frc2/command/CommandScheduler.cpp:222
#6  0x00007ffff7e979ea in frc::IterativeRobotBase::LoopFunc() (this=
    0x5555555894c0 <frc::impl::RunRobot<Robot>(std::mutex&, Robot**)::theRobot>)
    at ../wpilibc/src/main/native/cpp/IterativeRobotBase.cpp:165
#7  0x00007ffff7eb033c in std::function<void ()>::operator()() const (this=0x7fffe1d71800)
    at /usr/include/c++/10.2.0/bits/std_function.h:622
#8  frc::TimedRobot::StartCompetition()
    (this=this@entry=0x5555555894c0 <frc::impl::RunRobot<Robot>(std::mutex&, Robot**)::theRobot>)
    at ../wpilibc/src/main/native/cpp/TimedRobot.cpp:49
#9  0x00005555555697e9 in frc::impl::RunRobot<Robot>(std::mutex&, Robot**)
    (m=..., robot=robot@entry=0x55555558a6a0 <frc::StartRobot<Robot>()::robot>)
    at ../wpilibc/src/main/native/include/frc/RobotBase.h:33
#10 0x00005555555698b9 in frc::StartRobot<Robot>()::{lambda()#1}::operator()() const (this=<optimized out>)
    at ../wpilibc/src/main/native/include/frc/RobotBase.h:53
#11 std::__invoke_impl<void, frc::StartRobot<Robot>()::{lambda()#1}>(std::__invoke_other, frc::StartRobot<Robot>()::{lambda()#1}&&) (__f=<optimized out>) at /usr/include/c++/10.2.0/bits/invoke.h:60
#12 std::__invoke<frc::StartRobot<Robot>()::{lambda()#1}>(frc::StartRobot<Robot>()::{lambda()#1}&&)
    (__fn=<optimized out>) at /usr/include/c++/10.2.0/bits/invoke.h:95
#13 std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=<optimized out>) at /usr/include/c++/10.2.0/thread:264
#14 std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> >::operator()() (this=<optimized out>)
    at /usr/include/c++/10.2.0/thread:271
#15 std::thread::_State_impl<std::thread::_Invoker<std::tuple<frc::StartRobot<Robot>()::{lambda()#1}> > >::_M_run()
    (this=<optimized out>) at /usr/include/c++/10.2.0/thread:215
#16 0x00007ffff2d81c24 in std::execute_native_thread_routine(void*) (__p=0x555555af2000)
    at /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:80
#17 0x00007ffff2b393e9 in start_thread () at /usr/lib/libpthread.so.0
#18 0x00007ffff2a67293 in clone () at /usr/lib/libc.so.6
(gdb) 

```

PeterJohnson, 2021-01-20:
This looks like it's a pretty major problem with C++ new commands, and we'll likely find this bug elsewhere, and teams may easily run into it too.  The problem is that the constructor of SetDistanceToBox does a lambda capture of "this".  The AddCommands() line in Autonomous constructs a *temporary* SetDistanceToBox and a *copy* is made of this temporary into a std::unique_ptr.  After AddCommands() finishes, the temporary is destroyed and now the std::function's captured "this" pointer (which has been copied) is dangling.

closed 2021-01-21 by PeterJohnson

Modelmat, 2021-01-21:
Given that you stated that teams are likely to run into this elsewhere, would you like an issue tracking that here or on frc-docs?

PR #3111 (2021-01-21): [examples] Fix segfault in GearsBot C++ example
This is a temporary fix to prevent a dangling this pointer in the
DriveStraight and SetDistanceToBox commands by directly capturing the
drivetrain pointer by value instead.

Closes #3109