wpilibsuite/allwpilib #3120: Tools python scripts are broken

prensing, 2021-01-23:
**Describe the bug**
The ToolsUpdater.py script in 2021 is no longer working. It does not correctly find the WPILib Java installation.
This appears to be true of the other Python starter scripts which need Java (eg shuffleboard.py).

**To Reproduce**
Run "python tools/ToolsUpdater.py".  The program does not find any Java installation.

It is supposed to find the built-in java installation as the first option. However, what it tries for "jdk_dir" is "/home/ligerbots/wpilib/2021/tools/jdk/bin/java". It needs to strip off one more subdirectory before adding on the jdk ones.

**Expected behavior**
In the first try, "jdk_bin" should be "/home/ligerbots/wpilib/2021/jdk/bin/java", no "tools" directory in the path.

**Desktop (please complete the following information):**
 - OS: Linux

prateekma, 2021-01-23:
This was fixed in https://github.com/wpilibsuite/WPILibInstaller-Avalonia/commit/d8684840b23aabde1d31480eaa20ad875bb9349c.

Have you tried the 2021.2.1 installer?

prensing, 2021-01-23:
Umm. No because there was no announcement? You guys need to be clear about how to update. I just did the autoupdate inside VSCode. If that is not appropriate or sufficient, then you need to get the word out to use the full installer instead.

prensing, 2021-01-23:
Also, I fixed "ToolsUpdater.py" in 2021 (which started from the original 2021.1.2 version) and then ran it. It downloaded all the tools (did not have them before), but it downloaded broken Python scripts. 
Wherever it is getting those, that version needs to be updated.

prateekma, 2021-01-23:
You also need to fix ScriptBase.py with the changes from that commit.

ThadHouse, 2021-02-04:
This has been fixed for a few releases, so closing.

closed 2021-02-04 by ThadHouse