wpilibsuite/allwpilib #3132: Loop Overrun for Swerve Pose Estimator while calling addVisionMeasurement()

deslobodzian, 2021-01-27:
**Describe the bug**
While running the SwervePoseEstimator at 50hz, for the first 30-60 seconds a Loop Overrun of 0.03-0.07 seconds occur frequently assuming to be JVM warming up. Dies off a bit after receiving 0.021-0.06s loop overruns less frequently, roughly every 5-20 seconds. Also see very high CPU usage at startup.

**To Reproduce**
Steps to reproduce the behavior:
1. call addVisionMeasurement()

**Expected behavior**
Under 20ms(although sound unrealistic)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[no-visionUpdate.txt](https://github.com/wpilibsuite/allwpilib/files/5877334/no-visionUpdate.txt)
[visionUpdate enabled](https://youtu.be/nb92Qk9LtYc) (youtube video of log)

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Java version 11.0.8

calcmogul, 2021-01-27:
When an old vision measurement gets applied, it rewinds the state estimate and error covariance back to that vision measurement's timestamp, then replays all the motor inputs and local measurements that came after. That replay is probably taking a while. We should consider implementing https://www.researchgate.net/publication/3908304_The_Square-Root_Unscented_Kalman_Filter_for_State_and_Parameter-Estimation to make the predict-correct cycle faster.

calcmogul, 2022-08-15:
Closed by #4111.

closed 2022-08-15 by calcmogul