wpilibsuite/allwpilib #3136: SelectCommand not working when Object is a String

jasondaming, 2021-01-29:
When using SelectCommand with a String selector I get an error: `Supplier<String> cannot be converted to Supplier<Object>`

**To Reproduce**
Try to build this:
```
Supplier<String> i  = ()-> selectedPath;

// Select Command based on selectedPath from above
         new SelectCommand(Map.ofEntries(
            Map.entry("1R", 
               // Reset robot pose to the beginning of 1R and Run it
               m_robotDrive.createCommandForTrajectory(trajectory1R, true).withTimeout(15).withName("GalacticSearch1R")),
            Map.entry("1B", 
               // Reset robot pose to the beginning of 1B and Run it
               m_robotDrive.createCommandForTrajectory(trajectory1R, true).withTimeout(15).withName("GalacticSearch1B")),
            Map.entry("2R", 
               // Reset robot pose to the beginning of 2R and Run it
               m_robotDrive.createCommandForTrajectory(trajectory1R, true).withTimeout(15).withName("GalacticSearch2R")),
            Map.entry("2B", 
               // Reset robot pose to the beginning of 2B and Run it
               m_robotDrive.createCommandForTrajectory(trajectory1R, true).withTimeout(15).withName("GalacticSearch2B"))),
         i)```

**Desktop (please complete the following information):**
 - OS:  Windows
 - Java version WPILib 2.1

PeterJohnson, 2021-01-29:
What happens if you declare i as a `Supplier<Object>`?

jasondaming, 2021-01-29:
That did fix it.  The question is can we fix this to take a Supplier<String>?

KangarooKoala, 2023-08-24:
If we change the type parameters of the map and supplier from `Object` to `?` (the wildcard type parameter), then your example would work. We could also change `SelectCommand` to have a type parameter `T` (or `K` for key) and change the type parameters of the map and supplier to `? extends T`/`? extends K`, which would better match C++ and add more type safety, but would cause warnings about raw types for existing uses.

closed 2023-10-30 by PeterJohnson