wpilibsuite/allwpilib #3138: Add Trajectory + operator

dawonn, 2021-01-30:
I would like to concatenate multiple trajectories, but that's not possible with the current API so we need to use a state machine to execute multiple paths in a sequence. 

It would be grand if frc::Trajectory supported the + overload to concatenate trajectories, like std::string.

prateekma, 2021-01-30:
This sounds like a great idea!

Do you think we should print a warning if the user attempts to concatenate two trajectories where the end pose of the first doesn't match the start pose of the other?

dawonn, 2021-01-30:
Hmmm, my gut says no.

If the points are roughly in the same area the controller should still be
able to follow it... I think.

But how robust the controller is (or is not) should not put a constraint on
the trajectory data structure.

Starlight220, 2021-01-30:
I think that the second trajectory should be "rebased" (`relativeTo()`)  onto the end pose of the first trajectory. Unless that is precisely what we don't want to do because it changes the end pose of the total trajectory.

dawonn, 2021-01-30:
Proof of concept:

```
      wpi::SmallString<64> deployDirectory;
      frc::filesystem::GetDeployDirectory(deployDirectory);
      wpi::sys::path::append(deployDirectory, "output");
      wpi::sys::path::append(deployDirectory, "Bounce-#.wpilib.json");
      auto i = deployDirectory.rfind('#');

      // Load each path
      deployDirectory[i] = '1';
      auto t1 = frc::TrajectoryUtil::FromPathweaverJson(deployDirectory);
      auto s1 = t1.States();

      deployDirectory[i] = '2';
      auto t2 = frc::TrajectoryUtil::FromPathweaverJson(deployDirectory);
      auto s2 = t2.States();

      // Modify the time for each point in the second path
      auto start_time = s1.back().t + 10_ms;
      for (size_t i = 0; i < s2.size(); i++)
      {
        s2[i].t += s1.back().t + 10_ms;         
      } 

      // Add the second path to the overall path
      s1.insert( s1.end(), s2.begin(), s2.end() );

      // Save this Trajectory
      m_trajectory = frc::Trajectory(s1);
```

I did not set the last point of the first path and second point of the second path to the same point, just somewhat close-ish. I then just had to modify all the timestamps in the second path so they come. I tested the first path bing forward, and second path being reverse as well.... ;)

PR #3139 (2021-02-17): [wpimath] Add methods to concatenate trajectories
Closes #3138 

I'm not so sure about adding a seemingly random 10_ms (like the snippet in #3138) so I just discarded the first state in the second trajectory and added the total time of the first trajectory to the rest of the states.
closed 2021-02-17 by PeterJohnson