wpilibsuite/allwpilib #3146: Min acceleration greater than its max acceleration

alice15311, 2021-02-01:
We've been trying to get our robot to follow a trajectory for a few weeks now without much success. Today, we tried starting over from scratch and re-characterizing the robot. When we finished analyzing the data, we tried to have the robot follow the new trajectory only for it to fail with the error `The constraint's min acceleration is greater than its max acceleration`. We're not using a custom trajectory constraint, so we don't even know where the min acceleration is set. We tried increasing the `kMaxAccelerationMetersPerSecondSquared` constant, but made the error even more ambiguous, only saying something to the effect of "Something went wrong." We're at a loss of what the issue could be, so any help would be greatly appreciated.

[image]

prateekma, 2021-02-01:
@alice15311 So I played around with the code and basically what's happening is that given your characterization gains, it is not possible to follow certain parts of the trajectory without exceeding 10V. You can increase the threshold to 10.5V and everything should generate correctly.

calcmogul, 2021-02-01:
I wonder if we can have the voltage constraint detect this itself and give a more useful error.

samperlmutter, 2021-02-06:
@prateekma What did you do to determine that certain parts of the trajectory exceeded 10V?

prateekma, 2021-02-06:
The min and max acceleration values from the constraint didn't make sense with 10V but everything checked out with 10.5V. I agree with Tyler that we probably need a better error message for this.

PR #5492 (2023-08-01): [wpimath] Give infeasible trajectory constraints a better exception message
Fixes #3146.
commit 815a8403 (2023-08-01): [wpimath] Give infeasible trajectory constraints a better exception message (#5492)

Fixes #3146.
closed 2023-08-01 by PeterJohnson