wpilibsuite/allwpilib #3159: Make everything that needs to be closed AutoCloseable

Starlight220, 2021-02-08:
Regarding instructing on writing unit tests (wpilibsuite/frc-docs#1203).
Since that article will advise to close everything to provide a clean slate for each test, we can't have leaking resources and everything that should be closed should implement `AutoCloseable`. Most hardware classes already implement it, but none of the simulation stuff does. 
The `XxxSim` classes have a `resetData()` method. If it fits, they should implement `AutoCloseable` and have `close()` call `resetData()`. It should be very clear that if something needs to be closed/reset - it has `close()`. Teams won't go looking for `reset` functions, and it might also confuse with sensor resets. 
I'm not sure if controller/physics/etc need to be closed, same goes for NT stuff. 
@PeterJohnson - anything?

PeterJohnson, 2021-02-08:
The Sim classes generally do not need to be AutoCloseable, as there are no resources to clean up.  CallbackStore is the key exception and is already AutoCloseable.  Most likely resetData is headed for removal (or at least documented with more warning signs that it's for special case use only).

NT is a whole different challenge.  This might get easier if we go in a pub/sub direction.

Starlight220, 2022-05-02:
It seems that there isn't anything that needs to be AutoCloseable and isn't; closing.

closed 2022-05-02 by Starlight220