wpilibsuite/allwpilib #3165: Support for Windows ARM such as Microsoft Surface Pro

philltran, 2021-02-10:
We have a student with a Microsoft Surface Pro and discovered that the windows 64 bit installer of WPILib does not work on ARM based machines. 

We don't expect a build to be released this season, but am putting in the feature request now for next season.

Also, do you know if a guide exists for manually installing the dependencies on ARM based Windows?

Thank you.

Daltz333, 2021-02-10:
I believe that there are quite a bit of complications for building for ARM. @ThadHouse

ThadHouse, 2021-02-10:
Windows on Arm only supports 32 bit x86 apps, so you'd need to use the 32 bit installer.

As for fully supporting Arm, I've been looking into getting support. Currently the installer, java and the c++ compiler all have no way of being built natively for arm. Something that can be looked into, but not sure how feasible all that will be even within a year.

philltran, 2021-02-10:
Thank you for the updates.

ThadHouse, 2021-07-09:
Based on how gradle works, its actually unlikely that we will be able to ship a native Arm64 build of everything WPILib provides. However, the support for 64 bit executables coming in Windows 11 should make existing 64 bit builds work. However, this will likely require Windows 11, and I suspect we will not be able to get working Windows 10 support.

atilling, 2022-01-21:
Does the 2022 install fully function on a windows 11 ARM install?

ThadHouse, 2022-01-21:
Its never been tested, as none of us have a Windows 11 ARM device on the dev team. Theoretically the 64 bit installer should work, as Windows 11 now has 64 bit x86 emulation, but its never been tested.

Starlight220, 2022-08-29:
Any update on this? Can this be closed?

atilling, 2022-08-29:
I can state it worked for me in windows 11, but I was running windows 11 on an M1 Mac Book Prod with windows running on VMWare fusion preview edition

ThadHouse, 2022-08-29:
I won't want to close this until we have a native arm windows version. We've done it for mac, but doing it for windows is going to be harder.

auscompgeek, 2025-04-27:
Now that [GitHub Actions has Windows arm64 runners](https://github.blog/changelog/2025-04-14-windows-arm64-hosted-runners-now-available-in-public-preview/) this is slowly becoming a lot easier.

ThadHouse, 2025-04-27:
We actually build most things for Windows arm64 at this point. The main thing stopping this are Java tools. Its really hard to find a copy of JavaFX that works on Windows arm64. 

However, that problem will implicitly go away in 2027 when all the JavaFX tools are gone.

auscompgeek, 2026-09-04:
I think we can call this done now that we have a Windows ARM installer for 2027 alpha 7? The only thing that isn't native ARM is just the Systemcore C++ toolchain I think, but as mentioned in the other issue the x86-64 emulation should work.

ThadHouse, 2026-09-04:
Yeah we'll call this done for now.

closed 2026-09-04 by ThadHouse