wpilibsuite/allwpilib #3166: Simgui crashes simulation with SEGV in libhalsim_gui

chauser, 2021-02-11:
Desktop simulation suddenly stopped working, producing the following in the vscode terminal every time I tried to run it with the `libhalsimgui` enabled.
```
 cd /home/hauser/FRCProjects/autotest ; /usr/bin/env HALSIM_EXTENSIONS=/oldRoot/home/hauser/FRCProjects/autotest/build/tmp/expandedArchives/halsim_gui-2021.2.2-linuxx86-64.zip_15b82d10e73483ec37a38f1b5f97e77f/linux/x86-64/shared/libhalsim_gui.so: DYLD_LIBRARY_PATH=/oldRoot/home/hauser/FRCProjects/autotest/build/tmp/jniExtractDir LD_LIBRARY_PATH=/oldRoot/home/hauser/FRCProjects/autotest/build/tmp/jniExtractDir /oldRoot/home/hauser/wpilib/2021/jdk/bin/java -agentlib:jdwp=transport=dt_socket,server=n,suspend=y,address=localhost:35237 -Djava.library.path=/oldRoot/home/hauser/FRCProjects/autotest/build/tmp/jniExtractDir -Dfile.encoding=UTF-8 @/tmp/cp_73iwo5bgze6hhx4pfee8qkt0t.argfile frc.robot.Main 
HAL Extensions: Attempting to load: libhalsim_gui
Simulator GUI Initializing.
Simulator GUI Initialized!
HAL Extensions: Successfully loaded extension
********** Robot program starting **********
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f52be46768e, pid=5368, tid=5369
#
# JRE version: OpenJDK Runtime Environment AdoptOpenJDK (11.0.8+10) (build 11.0.8+10)
# Java VM: OpenJDK 64-Bit Server VM AdoptOpenJDK (11.0.8+10, mixed mode, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# C  [libhalsim_gui.so+0x8a68e]  (anonymous namespace)::KeyboardJoystick::Update()+0x19e
#
# Core dump will be written. Default location: Core dumps may be processed with "/usr/share/apport/apport %p %s %c %d %P %E" (or dumping to /oldRoot/home/hauser/FRCProjects/autotest/core.5368)
#
# An error report file with more information is saved as:
# /oldRoot/home/hauser/FRCProjects/autotest/hs_err_pid5368.log
#
# If you would like to submit a bug report, please visit:
#   https://github.com/AdoptOpenJDK/openjdk-support/issues
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
Aborted (core dumped)
```
Another issue report here led me to suspect `imgui.ini` so I moved it aside and let a new one be created. The simulation then started up fine. The broken `imgui.ini` is atttached here [imgui.ini-broken.txt](https://github.com/wpilibsuite/allwpilib/files/5963358/imgui.ini-broken.txt).

This is with the latest wpilib on Ubuntu 18.04. To the best of my knowledge I have not configured or used any keyboard joysticks.

PeterJohnson, 2021-02-11:
Thanks for the ini file, I've reproduced the crash and am investigating.

chauser, 2021-02-11:
Hi Peter. I see you've already created a pull request for a fix. I am
enormously impressed by the dedication of you and the wpilib team -- issues
are addressed within hours and often fixed just as quickly. Thanks for your
great support of FRC!

PR #3167 (2021-02-13): [sim] GUI: Make keyboard settings loading more robust
Check values during load and operation to avoid potential crashes due to
ini file errors or corruption.

Fixes #3166.
closed 2021-02-13 by PeterJohnson