wpilibsuite/allwpilib #3168: ControllerUtil.getModulusError doesn't always return the closest error value

jsimpso81, 2021-02-11:
**Describe the bug**
ControllerUtil.getModulusError doesn't always return the closest error value.  It seems that this routine may soon be obsolete, but I thought I'd send the information anyway.  I only tried the java version.

I quickly tried the inputModulus routine which seems like the replacement.  If this is used to get an error then the limits on the SP-PV error seem like they should be +/- (max-min)*0.5.  The range of the error is potentially different than the range of the input.  Therefore there may be an issue here too.

The test program below seemed to show the following issues:
================= ORIGINAL getModulusError ROUTINE =======================
11 - Should equal 170.0 = -190.0
13 - Should equal 170.0 = -190.0
================= ORIGINAL inputModulus ROUTINE =======================
 7 - Should equal -20.0 = 340.0
 8 - Should equal -20.0 = 340.0
 9 - Should equal -20.0 = 340.0

**To Reproduce**
Here is a sample test program that shows the issue

//  TRY OUT MODULUS ERROR CALC.
package frc.robot.test;
import edu.wpi.first.wpilibj.controller.ControllerUtil;
import edu.wpi.first.wpiutil.math.MathUtil;

public final class TEST_MODULUS_ERROR_CALC {
    public static void main(String... args) {

        // ========================== TEST ORIGINAL
        System.out.println("================= ORIGINAL getModulusError ROUTINE =======================");
        // Test symmetric range
        System.out.println(" 1 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( ControllerUtil.getModulusError(170.0,         -170.0,         -180.0, 180.0) ) );
        System.out.println(" 2 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( ControllerUtil.getModulusError(170.0 + 360.0, -170.0,         -180.0, 180.0) ) );
        System.out.println(" 3 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( ControllerUtil.getModulusError(170.0,         -170.0 + 360.0, -180.0, 180.0) ) );
        System.out.println(" 4 - Should equal " + Double.toString(20.0D)+" = "+Double.toString( ControllerUtil.getModulusError(-170.0,          170.0,         -180.0, 180.0) ) );
        System.out.println(" 5 - Should equal " + Double.toString(20.0D)+" = "+Double.toString( ControllerUtil.getModulusError(-170.0 + 360.0, 170.0, -180.0, 180.0)));
        System.out.println(" 6 - Should equal " + Double.toString(20.0D)+" = "+Double.toString( ControllerUtil.getModulusError(-170.0, 170.0 + 360.0, -180.0, 180.0)));

        // Test range start at zero
        System.out.println(" 7 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( ControllerUtil.getModulusError(170.0, 190.0, 0.0, 360.0)));
        System.out.println(" 8 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( ControllerUtil.getModulusError(170.0 + 360.0, 190.0, 0.0, 360.0)));
        System.out.println(" 9 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( ControllerUtil.getModulusError(170.0, 190.0 + 360, 0.0, 360.0)));
    
        // Test asymmetric range that doesn't start at zero
        System.out.println("10 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( ControllerUtil.getModulusError(170.0, -170.0, -170.0, 190.0)));

        // JAS test
        System.out.println("11 - Should equal " + Double.toString(170.0D) +" = "+Double.toString( ControllerUtil.getModulusError(-190.0,     0.0,   10.0, 370.0)));
        System.out.println("12 - Should equal " + Double.toString(-110.0D)+" = "+Double.toString( ControllerUtil.getModulusError(-130.0,   -20.0, -180.0, 180.0)));
        System.out.println("13 - Should equal " + Double.toString( 170.0D) +" = "+Double.toString( ControllerUtil.getModulusError(   0.0,  190.0,   10.0, 370.0)));
        
        // ========================== TEST NEW ORIGINAL
        System.out.println("================= ORIGINAL inputModulus ROUTINE =======================");
        // Test symmetric range
        System.out.println(" 1 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( MathUtil.inputModulus(170.0 - (         -170.0),         -180.0, 180.0) ) );
        System.out.println(" 2 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( MathUtil.inputModulus(170.0 + 360.0 - ( -170.0),         -180.0, 180.0) ) );
        System.out.println(" 3 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( MathUtil.inputModulus(170.0 - (         -170.0 + 360.0), -180.0, 180.0) ) );
        System.out.println(" 4 - Should equal " + Double.toString(20.0D)+" = "+Double.toString( MathUtil.inputModulus(-170.0 - (          170.0),         -180.0, 180.0) ) );
        System.out.println(" 5 - Should equal " + Double.toString(20.0D)+" = "+Double.toString( MathUtil.inputModulus(-170.0 + 360.0 - ( 170.0), -180.0, 180.0)));
        System.out.println(" 6 - Should equal " + Double.toString(20.0D)+" = "+Double.toString( MathUtil.inputModulus(-170.0 - ( 170.0 + 360.0), -180.0, 180.0)));

        // Test range start at zero
        System.out.println(" 7 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( MathUtil.inputModulus(170.0 - ( 190.0), 0.0, 360.0)));
        System.out.println(" 8 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( MathUtil.inputModulus(170.0 + 360.0 - ( 190.0), 0.0, 360.0)));
        System.out.println(" 9 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( MathUtil.inputModulus(170.0 - ( 190.0 + 360), 0.0, 360.0)));
    
        // Test asymmetric range that doesn't start at zero
        System.out.println("10 - Should equal " + Double.toString(-20.0D)+" = "+Double.toString( MathUtil.inputModulus(170.0 - ( -170.0), -170.0, 190.0)));

        // JAS test
        System.out.println("11 - Should equal " + Double.toString(170.0D) +" = "+Double.toString( MathUtil.inputModulus(-190.0 - (     0.0),   10.0, 370.0)));
        System.out.println("12 - Should equal " + Double.toString(-110.0D)+" = "+Double.toString( MathUtil.inputModulus(-130.0 - (   -20.0), -180.0, 180.0)));
        System.out.println("13 - Should equal " + Double.toString( 170.0D) +" = "+Double.toString( MathUtil.inputModulus(   0.0 - (  190.0),   10.0, 370.0)));
[... 95 lines cut ...]
11 - Should equal 170.0 = 170.0
12 - Should equal -110.0 = -110.0
13 - Should equal 170.0 = 170.0
================= POTENTIAL REVISED ROUTINE =======================
 1 - Should equal -20.0 = -20.0
 2 - Should equal -20.0 = -20.0
 3 - Should equal -20.0 = -20.0
 4 - Should equal 20.0 = 20.0
 5 - Should equal 20.0 = 20.0
 6 - Should equal 20.0 = 20.0
 7 - Should equal -20.0 = -20.0
 8 - Should equal -20.0 = -20.0
 9 - Should equal -20.0 = -20.0
10 - Should equal -20.0 = -20.0
11 - Should equal 170.0 = 170.0
12 - Should equal -110.0 = -110.0
13 - Should equal 170.0 = 170.0

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Java version - not certain
 - I think I'm using the latest WPILIB release (I checked for an update before running test program.)

**Additional context**
Crude potential fix is included in the test program.

calcmogul, 2021-02-11:
Oof. Looks like we forgot to remove Java's ControllerUtil.getModulusError() before kickoff (it was added in March 2020, so it never saw a season). We already did so for C++.

jsimpso81, 2021-02-11:
Is there an issue with the newer inputModulus routine too? Or maybe it how it is used..  (Or maybe I just overlooked something.....)

calcmogul, 2021-02-11:
Here's how PIDController uses it:
```java
    if (m_continuous) {
      positionError =
          MathUtil.inputModulus(m_setpoint - m_measurement, m_minimumInput, m_maximumInput);
    } else {
      positionError = m_setpoint - m_measurement;
    }
```

calcmogul, 2021-02-11:
Here's the unit tests for it. Is there a missing scenario? https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/test/java/edu/wpi/first/wpiutil/math/MathUtilTest.java

jsimpso81, 2021-02-11:
So if the range is 0 - 360, shouldn't the error for the PID still be in the range of +/- 180?

calcmogul, 2021-02-11:
The inputs to inputModulus() need to all be errors. Looks like we're doing this wrong in PIDController.

calcmogul, 2021-02-11:
The correct usage is probably this, as per your initial post:
```java
    if (m_continuous) {
      positionError =
          MathUtil.inputModulus(m_setpoint - m_measurement,
                                -(m_maximumInput - m_minimumInput) / 2.0,
                                (m_maximumInput - m_minimumInput) / 2.0);
    } else {
      positionError = m_setpoint - m_measurement;
    }
```

jsimpso81, 2021-02-11:
I think that makes sense.  Thanks for the attention...

jsimpso81, 2021-02-11:
I think posts are passing quickly.  My last comment is out of order...  I didn't look at the latest code snipet..  Maybe that looks good.  My brain doesn't work quite that fast.  (I'd put the calc outside of the call so it only has to be done once.)

PR #3170 (2021-05-22): [wpilib] Fix PIDController continuous range error calculations
The inputs should all be errors, so the range should be symmetric.

Fixes #3168.
commit 87384ea6 (2021-05-22): [wpilib] Fix PIDController continuous range error calculations (#3170)

The inputs should all be errors, so the range should be symmetric.

Fixes #3168.
Fixes #3304.
closed 2021-05-22 by PeterJohnson